nRF24/RF24Network #154: Data getting lost

marcus011, 2020-02-27:
Hi @TMRh20 ,
As per suggest by you I have setup my devices as per reference of the image link you shared
`https://tmrh20.github.io/RF24Network/Tuning.html`
So I have created network like this so that I can transmit data for long distance.

> 01111 is Sending Device
> 01112 is the gateway
>  Rest of them broadcasting the data 
>  01111 -> 0111 -> 011 -> 01 -> 00 -> 02 -> 012 -> 0112 -> 01112 

> Settings Used in all devices
> 
> DataRate : 250kbps
> RF Power : HIGH
> PAYLOAD SIZE : 14
> The sender & broadcast devices are Arduino based & the end device is Raspberry pi using c++.

> The Broadcast Device is only having code with network.update() in while loop

The data is successfully transmitting to the end device & back to Sender,
But only 7 out of 10 times its reaching successfully, sometimes it transmit through each devices but at the end device it does not reach. I have also connected a 10uf Capacitor on top of it & right now all devices are closed to each other as I am testing.

So please let me know how I can send my data efficiently , Because these data need to be send when any emergency happen.

Thanks

TMRh20, 2020-02-27:
To start off,  a 70% success rate might not be too bad considering transmission is taking place over 8 hops or 8 individual transmissions.

Depending on the exact situation, you might increase the success rate a bit by spacing the devices an appropriate distance apart, or using a lower power level in testing. 

You also need to consider that wireless communication especially in the 2.4Ghz spectrum is not guaranteed and can be susceptible to interference etc.  With that said, I would suggest using a frequency in the lower end of its range (like radio channel 1) as lower frequencies work better over distances and through objects.

In addition, RF24Network can use network-acks, so you can have the sending device detect if a payload was sent OK. In this case you could attempt re-transmission at random intervals until successful or for a set amount of time. The downside is that this can result in duplicate payloads, but they can be easily detected and discarded by recipient.

You can also adjust some hard coded values here https://github.com/nRF24/RF24Network/blob/master/RF24Network.cpp#L107

The retries can be changed from 5 to 15, and with that, you can adjust the txTimeout value to something like 50 or 75. This will make the radios automatically re-attempt each send up to 15 times instead of 5.

Hope this helps,

TMRh20

marcus011, 2020-02-28:
Thanks @TMRh20  for suggestion , I will try today & let tpu know further.
One more thing i need to know ,is it possible we can disable Acknowledgement handling means just push the data & don't care about ACK

marcus011, 2020-02-28:
One more thing i want to know about
  ```
radio.setAutoack();
radio.enableDynamicAck();
  radio.enableDynamicPayloads()
```
I have gone through your documentation but I not able to understand when to use these,will enabling or disabling this affect my data transfer or not.

So please let me know when these 3 api's are useful to use & when not.
Thanks

marcus011, 2020-02-28:
@TMRh20 
As per your suggestion ,I have changed my channel to 1  ,adjust txtimeout & done some tweeks in network.update function & added my own code if transmission failed for next node then retransmit after every second till a limit,if fail then create packet & tell the sender node that next node is inactive. It working fine now,all thanks to you.

One more thing i need to ask when data successfully reached the end node ,the end node transmit 193 code back , is it handle automatically & can we disable that ?  will it affect something?

Thanks

TMRh20, 2020-02-28:
To answer your first and last questions, there are two types of acknowledgement, network level and radio level. 

To disable acks at network level, use message types 1 through 64. This prevents the end node from sending back code 193 (network ack)

To disable acks at radio level, refer to the same hardcoded values where the txTimeout value is found, and set the radio.setRetries(0,0); This is not usually recommended.

The three apis you mention should be left as-is.

marcus011, 2020-02-28:
Thanks @TMRh20  for your help, help me alot.

closed 2020-02-28 by marcus011