nRF24/RF24Network #155: Esp32 with rf24-network

itriedsohar, 2020-03-04:
Hello. I have a simple question. Is it possible to use the rf24-network library with the esp32 module. I have searched all over and have not got help. Please help me

TMRh20, 2020-03-04:
Yes

itriedsohar, 2020-03-04:
@TMRh20 I was trying to use it on my arduino ide but it brings a warning saying that the library is not supported on the esp32 architectecture. I am trying to use the esp32 as a base node receiving values from 4 sensor nodes. Before uploading, that particular warning shows up, and after uploading, the serial monitor does not show any values. I know the code is correct because when I test it with an arduino uno it works just fine. But I need the esp32 for iot capacity. Is there anything I am supposed to add or remove. Thank you.

TMRh20, 2020-03-04:
Take a look at this thread https://github.com/nRF24/RF24/issues/393 for correct wiring etc.

You basically want to do standard troubleshooting, so I would suggest testing with the core RF24 driver first, and make sure you are getting correct output from the included printDetails(); function.

Once that is ok, then move to RF24Network and you should have better luck.

TMRh20, 2020-03-05:
> @TMRh20 I used the same module with the RF24 library with this sketch https://howtomechatronics.com/tutorials/arduino/arduino-wireless-communication-nrf24l01-tutorial/ and it worked just fine (receiving "HELLO WORLD" string). I also used your recomend forum nRF24/RF24#393) to acquire the correct pin connection between the esp32 and nrf24l01. It's only when I use the RF24-NETWORK library that the warning appears. The code actually uploads but nothing is shown on the serial monitor. I will try harder n get back to you if I fail or succeed but any new help would really be grateful. Thanks again

We need an accurate idea of what is happening to respond. 

When you say it shows nothing over serial, is that like completely nothing, not even RF24Network/examples/helloworld_rx/");   or are you just not getting a response?

itriedsohar, 2020-03-05:
@TMRh20 @Avamander managed to figure it out. Its embaracing but I just didn't use the right pins in the radio object declaration. But thanks so much. U guyz rock

closed 2020-05-27 by TMRh20