nRF24/RF24Network #156: No Transmission when Antenna Attached

marcus011, 2020-04-08:
So Today I was testing my nodes & uploaded example code from the library & just added some code so that when one node RX data it will write back to the sender, so after testing I found out that when I have attached the antenna to NRF24l01, data is not able to transmit back but data is Receiving, so when I remove the antenna it start transmitting but when putting back on its not transmitting this problem occurs when using PA level MAX or HIGH & under LOW MIN it's working well, and sometimes holding the antenna with hand transmits the data ( weird ).

So can anyone tell whats the issue going on & how to resolve for MAX & HIGH level?

> Settings on NRF
> channel 90
> radio.setDataRate(RF24_250KBPS);
> radio.setPALevel(RF24_PA_MAX);
> Power to NRF through AMS1117 Externally with 10uf Capacitor

g0uwz, 2020-04-08:
Hi are you testing these on the bench together in very close proximity ? 
Sounds to me that the tx is too strong and overloading the receiver. Try increasing distance between the units. 
Holding the antenna with your hand is reducing the rf level sent.

marcus011, 2020-04-09:
OK I will try to increase the distance ,right now it was around ~4 feet
Will let you know if it works

marcus011, 2020-04-09:
@g0uwz Results were same no effect, but holding it again works but without holding its not working (only recieving not transmitting)

g0uwz, 2020-04-09:
Strange, it sounded like a classic case of rf swamping. 
Sorry i can't help, but will be interested to hear the outcome.

TMRh20, 2020-04-09:
Still sounds like rf swamping to me. I would try the old wrapping the
module in tinfoil or put in metal box trick.

https://hackaday.com/2016/05/31/fixing-the-terrible-range-of-your-cheap-nrf24l01-palna-module/

https://forum.mysensors.org/topic/1877/solved-2-x-nrf24l01-pa-lna-with-rf24_pa_max/10
https://forum.arduino.cc/index.php?topic=382021.0

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

marcus011, 2020-04-09:
Thanks guys, I will give it a try & let you know it's result.

g0uwz, 2020-04-11:
Hi 
Strange i am sat scratching my head, working with a couple of rfnano's.
The distance between units can not go beyond 900mm. 
I have changed data rate and tx power setting but only improves slightly on max power. 
So with this in mind, i have been looking at the nordic schematic in comparison to other manufacturers and noticed on mine a 1pf capcitor is missing and the track goes direst to ground (on the cold side of the dc blocking cap from on board antenna.
You may be wise to try a different brand of tx, again it sounds like it's too strong to me.
i am just going to flash these nanos, and do a distance test.

closed 2020-08-16 by TMRh20