nRF24/RF24Network #158: Can't send message to Node 014!

adamzxtan, 2020-07-20:
So in my sensor network, I have setup 6 nodes of slaves with the address of 01, 02, 03, 04, 05, 014. However, no matter what I do the master node will not be able to return success upon sending to 014 node. I changed the address to 015 still the same result. Here's the debugging message from serial monitor:

**Master node:**
``
11: NET Pipe 5 on node 00 has address cccccccce3
11: NET Pipe 4 on node 00 has address cccccccc3e
14: NET Pipe 3 on node 00 has address ccccccccce
18: NET Pipe 2 on node 00 has address cccccccc33
22: NET Pipe 1 on node 00 has address cccccccc3c
26: NET Pipe 0 on node 00 has address ccccccccc3

STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccc3ee3
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x7d
RF_SETUP	 = 0x27
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

2643: NET Sending id 1 from 00 to 01 type 84
2644: NET message 00 00 00 c8 
2646: MAC Sending to 01 via 01 on pipe 5
2649: NET Pipe 5 on node 01 has address cccccc3ce3
3732: NET Sending id 2 from 00 to 02 type 84
3732: NET message 00 00 00 c8 
3733: MAC Sending to 02 via 02 on pipe 5
3736: NET Pipe 5 on node 02 has address cccccc33e3
4822: NET Sending id 3 from 00 to 03 type 84
4823: NET message 00 00 00 c8 
4824: MAC Sending to 03 via 03 on pipe 5
4827: NET Pipe 5 on node 03 has address cccccccee3
5910: NET Sending id 4 from 00 to 04 type 84
5910: NET message 00 00 00 c8 
5911: MAC Sending to 04 via 04 on pipe 5
5914: NET Pipe 5 on node 04 has address cccccc3ee3
6998: NET Sending id 5 from 00 to 05 type 84
6999: NET message 00 00 00 c8 
6999: MAC Sending to 05 via 05 on pipe 5
7003: NET Pipe 5 on node 05 has address cccccce3e3
8086: NET Sending id 6 from 00 to 014 type 84
8086: NET message 00 00 00 c8 
8087: MAC Sending to 014 via 04 on pipe 5
8090: NET Pipe 5 on node 04 has address cccccc3ee3
8150: NET Sending id 7 from 00 to 014 type 84
8151: NET message 00 00 00 c8 
8151: MAC Sending to 014 via 04 on pipe 5
8155: NET Pipe 5 on node 04 has address cccccc3ee3
8212: NET Sending id 7 from 00 to 014 type 84
8213: NET message 00 00 00 c8 
8213: MAC Sending to 014 via 04 on pipe 5
8217: NET Pipe 5 on node 04 has address cccccc3ee3
8451: NET Sending id 7 from 00 to 014 type 84
8452: NET message 00 00 00 c8 
8452: MAC Sending to 014 via 04 on pipe 5
8456: NET Pipe 5 on node 04 has address cccccc3ee3
8689: NET Sending id 7 from 00 to 014 type 84
8690: NET message 00 00 00 c8 
8690: MAC Sending to 014 via 04 on pipe 5
8694: NET Pipe 5 on node 04 has address cccccc3ee3
8928: NET Sending id 7 from 00 to 014 type 84
8929: NET message 00 00 00 c8 
8929: MAC Sending to 014 via 04 on pipe 5
8933: NET Pipe 5 on node 04 has address cccccc3ee3
9166: NET Sending id 7 from 00 to 014 type 84
9167: NET message 00 00 00 c8 
9167: MAC Sending to 014 via 04 on pipe 5
9171: NET Pipe 5 on node 04 has address cccccc3ee3
9405: NET Sending id 7 from 00 to 014 type 84
9406: NET message 00 00 00 c8 
9406: MAC Sending to 014 via 04 on pipe 5
9410: NET Pipe 5 on node 04 has address cccccc3ee3
9644: NET Sending id 7 from 00 to 014 type 84
9645: NET message 00 00 00 c8 
9645: MAC Sending to 014 via 04 on pipe 5
9649: NET Pipe 5 on node 04 has address cccccc3ee3
9882: NET Sending id 7 from 00 to 014 type 84
9883: NET message 00 00 00 c8 
9883: MAC Sending to 014 via 04 on pipe 5
9887: NET Pipe 5 on node 04 has address cccccc3ee3
10121: NET Sending id 7 from 00 to 014 type 84
10122: NET message 00 00 00 c8 
10123: MAC Sending to 014 via 04 on pipe 5
10126: NET Pipe 5 on node 04 has address cccccc3ee3
10360: NET Sending id 7 from 00 to 014 type 84
10360: NET message 00 00 00 c8 
10361: MAC Sending to 014 via 04 on pipe 5
10365: NET Pipe 5 on node 04 has address cccccc3ee3
11640: MAC Received on 4 id 7 from 00 to 00 type 193
11640: NET message 00c8
11674: MAC Received on 4 id 7 from 00 to 00 type 193
11675: NET message 00c8
11676: MAC Received on 4 id 7 from 00 to 00 type 193
11680: NET message 00c8
``

**Slave node 014:**
``
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xcccccc33cc 0xcccc3c3e3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x7d
RF_SETUP	 = 0x27
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

13881: MAC Received on 5 id 6 from 00 to 014 type 84

13882: NET message 00c8

13882: NET Enqueue @0 Received packet!
13886: NET message size 4
13888: NET r message 00 00 00 c8 

13930: MAC Received on 5 id 7 from 00 to 014 type 84

13931: NET message 00c8

13931: NET Enqueue @0 13933: NET message size 4
13935: NET r message 00 00 00 c8 

13978: MAC Received on 5 id 7 from 00 to 014 type 84

13979: NET message 00c8

13980: NET Enqueue @0 13982: NET message size 4
13984: NET r message 00 00 00 c8 

14215: MAC Received on 5 id 7 from 00 to 014 type 84

14216: NET message 00c8

14217: NET Enqueue @0 Received packet!
14220: NET message size 4
14222: NET r message 00 00 00 c8 

14454: MAC Received on 5 id 7 from 00 to 014 type 84

14454: NET message 00c8

14455: NET Enqueue @0 Received packet!
14458: NET message size 4
14460: NET r message 00 00 00 c8 

14695: MAC Received on 5 id 7 from 00 to 014 type 84

14696: NET message 00c8

14696: NET Enqueue @0 Received packet!
14700: NET message size 4
14702: NET r message 00 00 00 c8 

14934: MAC Received on 5 id 7 from 00 to 014 type 84

14935: NET message 00c8

14935: NET Enqueue @0 Received packet!
14939: NET message size 4
14941: NET r message 00 00 00 c8 

15170: MAC Received on 5 id 7 from 00 to 014 type 84

15171: NET message 00c8

15171: NET Enqueue @0 Received packet!
15175: NET message size 4
15177: NET r message 00 00 00 c8 

15416: MAC Received on 5 id 7 from 00 to 014 type 84

15417: NET message 00c8

15417: NET Enqueue @0 Received packet!
15421: NET message size 4
15423: NET r message 00 00 00 c8 

15652: MAC Received on 5 id 7 from 00 to 014 type 84

15653: NET message 00c8

15653: NET Enqueue @0 Received packet!
15657: NET message size 4
15660: NET r message 00 00 00 c8 

15894: MAC Received on 5 id 7 from 00 to 014 type 84

15894: NET message 00c8

15895: NET Enqueue @0 Received packet!
15898: NET message size 4
15900: NET r message 00 00 00 c8 

16131: MAC Received on 5 id 7 from 00 to 014 type 84

16132: NET message 00c8

16132: NET Enqueue @0 Received packet!
16135: NET message size 4
16138: NET r message 00 00 00 c8 
``

Obviously the 014 node is receiving message from the master node. But the master node doesn't seems like receive the network ack. Any idea on why this happens?

TMRh20, 2020-07-20:
Well surprisingly, I'm getting similar results. Not sure what or when this became an issue, but a small change will work around it for now:
1. Edit RF24Network.cpp here: https://github.com/nRF24/RF24Network/blob/master/RF24Network.cpp#L932 , add `delay(2);`
2. Recompile and upload to node 04 (the routing node for 014) only. You may have to switch boards in the IDE to force a full recompilation.  
  
The delay could be made higher if not working. I'm not sure why it needs to be there exactly, but seems to do the trick. Please let me know if this works, I'm going to take a closer look as soon as I get a chance.

adamzxtan, 2020-07-21:
I've changed the air data rate from 250kbps to 1mbps and now it works again. I'm not entirely sure why, but seems like I've ran into one of the common issue? https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md

Anyway I'll try out the suggestion, but just to clarify do I add the ```delay(2)``` before or after?
```/**Write it*/
  ok=write_to_pipe(conversion.send_node, conversion.send_pipe, conversion.multicast);
  delay(2);
```
or
```
/**Write it*/
  delay(2);
  ok=write_to_pipe(conversion.send_node, conversion.send_pipe, conversion.multicast);
```

TMRh20, 2020-07-21:
This isn't a problem with the build-in radio ACKs, its a RF24Network specific issue with software based ACKs.

I used a PA+LNA module also for the routing node in my test, not sure if it makes a difference yet.

The delay should go before the write. I've been using the following code to only delay when its a software
acked payload.
```
  if( directTo == TX_ROUTED && conversion.send_node == to_node && isAckType){   //<-- Add
    delay(2);                                                                   //<-- These
  }                                                                             //<-- Lines
  ok=write_to_pipe(conversion.send_node, conversion.send_pipe, conversion.multicast);  	
```

TMRh20, 2020-07-21:
FYI some further results on `Arduino Mega 2560 & PA+LNA` module, `Arduino Due & No Antenna module` , 1Mbps:
```
1. With Mega as 044 and Due as 04 routing node, the delay is not required
2. With Due as 044 and Mega as 04 routing node, the delay is required
3. Configuration of MIN, LOW power etc does not make a difference
```
The Due MCU is much faster and should be responding quicker to requests, but the Mega as routing node has the problems. I have to attribute this to the PA+LNA high powered modules behaving differently. It also explains why this has not been a bigger issue.

adamzxtan, 2020-07-25:
> This isn't a problem with the build-in radio ACKs, its a RF24Network specific issue with software based ACKs.
>
> I used a PA+LNA module also for the routing node in my test, not sure if it makes a [...]

This has been working well for me so far. I'm bringing out my setup to wide area test later tomorrow to see if it solves my issue at all. Also will be testing with 8 slaves sensor.

adamzxtan, 2020-08-11:
Hi @TMRh20 everything has been working fine so far. Other than some intermittent connection lost issue, or random missing ack packet, things are ok. So I'm closing this issue now. Thanks for the help.

closed 2020-08-11 by adamzxtan