nRF24/RF24Network #160: Can't reach the child on level 3

diygoodies, 2020-09-17:
Hi I made four absolutely similar boards with stm32 bluepill and nrf24l01 onboard. And on this moment I want allign them in line and to make connection from  base 00, thry child 001, then thry child 011 up to child 0111. Now I have sucsessfuly reached only level 2 chlild such as 011 and 021, level 3 and a child 0111 or 0211 stays unreached for now. The level 3 board is absolutly workable     I chechked it on level 2. Here is my code examples. Please give me a advise what can be wrong.

Base
`
/*
 Copyright (C) 2012 James Coliz, Jr. <maniacbug@ymail.com>

 This program is free software; you can redistribute it and/or
 modify it under the terms of the GNU General Public License
 version 2 as published by the Free Software Foundation.
 
 Update 2014 - TMRh20
 */

/**
 * Simplest possible example of using RF24Network 
 *
 * TRANSMITTER NODE
 * Every 2 seconds, send a payload to the receiver node.
 */

#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

#define RED_led PA8
#define GRN_led PA9
#define BLU_led PA10
#define BAT_sens PA3
#define RFM_enbl PC14
#define UID_BASE 0x1FFFF7E8U                  // Unique device ID register base address
RF24 radio(PA0, PA4); // CE, CSN

RF24Network network(radio);          // Network uses that radio

uint32_t UnitID_Part_1; 
uint32_t UnitID_Part_2;
uint32_t UnitID_Part_3;

uint16_t this_node = 00;  // Address of our node in Octal format
uint16_t node_array[4]={0,01,011,0111};  // Address of our node in Octal format
uint16_t other_node = 01;       // Address of the other node in Octal format
uint16_t node_total_count = 03; // Node total count in Octal format
uint16_t led_blink_count = 00; // Led blink count in Octal format

const unsigned long interval = 1500; //ms  // How often to send 'hello world to the other unit

unsigned long last_sent;             // When did we last send?
unsigned long packets_sent;          // How many have we sent already

struct payload_t {                  // Structure of our payload
  unsigned long ms;
  unsigned long counter;
};

void setup(void)
{
  Serial.begin(115200);
  UnitID_Part_1 = *(uint32_t*)(UID_BASE);
  UnitID_Part_2 = *(uint32_t*)(UID_BASE+0x04);
  UnitID_Part_3 = *(uint32_t*)(UID_BASE+0x08);

  
  if (UnitID_Part_1==0x66dff56)
  {
    this_node = 00;
  }

  //delay(5000);
  Serial.println("Start ");
  Serial.println("UID [HEX]  : "+String(UnitID_Part_1, HEX)+" "+String(UnitID_Part_2, HEX)+" "+String(UnitID_Part_3, HEX));
  
  Serial.println("RF24Network/examples/helloworld_tx/");

  pinMode(RFM_enbl, OUTPUT);
  digitalWrite(RFM_enbl, HIGH);
  pinMode(RED_led, OUTPUT_OPEN_DRAIN);
  pinMode(GRN_led, OUTPUT_OPEN_DRAIN);
  pinMode(BLU_led, OUTPUT_OPEN_DRAIN);
  pinMode(BAT_sens, INPUT);
  digitalWrite(RED_led, HIGH);
  digitalWrite(GRN_led, HIGH);
  digitalWrite(BLU_led, HIGH);
 
  SPI.begin();
  radio.begin();
  network.begin(/*channel*/ 90, /*node address*/ this_node);
}

void loop() {
  
  network.update();                          // Check the network regularly

  
  unsigned long now = millis();              // If it's time to send a message, send it!
  if ( now - last_sent >= interval  )
  {
    last_sent = now;

    Serial.print("Sending to ");
    Serial.println(node_array[other_node],DEC);
    payload_t payload = { millis(), packets_sent++ };
    RF24NetworkHeader header(/*to node*/ node_array[other_node]);
    bool ok = network.write(header,&payload,sizeof(payload));
    led_blink_count=other_node+1;
    if (ok)
    {
      Serial.println("ok.");
      while (led_blink_count!=0)
      {
        digitalWrite(GRN_led, LOW);
        delay(100);
        digitalWrite(GRN_led, HIGH);
        delay(100);
        led_blink_count--;
      }
    }
    else
    {
      Serial.println("failed.");
      while (led_blink_count!=0)
      {
        digitalWrite(RED_led, LOW);
        delay(100);
        digitalWrite(RED_led, HIGH);
        delay(100);
        led_blink_count--;
      }
    }
    if (other_node<node_total_count)
    {
     other_node++;
    }
    else
    {
      other_node=01;
    }
  }
}`

Child
`/*
 Copyright (C) 2012 James Coliz, Jr. <maniacbug@ymail.com>

 This program is free software; you can redistribute it and/or
 modify it under the terms of the GNU General Public License
 version 2 as published by the Free Software Foundation.
 
 Update 2014 - TMRh20
 */

/**
 * Simplest possible example of using RF24Network,
 *
 * RECEIVER NODE
 * Listens for messages from the transmitter and prints them out.
 */

#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

#define RED_led PA8
#define GRN_led PA9
#define BLU_led PA10
#define BAT_sens PA3
#define RFM_enbl PC14
#define UID_BASE 0x1FFFF7E8U
RF24 radio(PA0, PA4); // CE, CSN

RF24Network network(radio);      // Network uses that radio

uint32_t UnitID_Part_1; 
uint32_t UnitID_Part_2;
uint32_t UnitID_Part_3;

uint16_t this_node = 01;    // Address of our node in Octal format ( 04,031, etc)
uint16_t other_node = 00;   // Address of the other node in Octal format

struct payload_t {                 // Structure of our payload
  unsigned long ms;
  unsigned long counter;
};

void setup(void)
{
  Serial.begin(115200);
  UnitID_Part_1 = *(uint32_t*)(UID_BASE);
  UnitID_Part_2 = *(uint32_t*)(UID_BASE+0x04);
  UnitID_Part_3 = *(uint32_t*)(UID_BASE+0x08);

  if (UnitID_Part_1==0x66aff55)
  {
    this_node = 01;
  }
  if (UnitID_Part_1==0x66bff52)
  {
    this_node = 011;
  }
  if (UnitID_Part_1==0x674ff55)
  {
    this_node = 0111;
  }
  //delay(5000);
  Serial.print("Start ");
  Serial.println(this_node);
  Serial.println("UID [HEX]  : "+String(UnitID_Part_1, HEX)+" "+String(UnitID_Part_2, HEX)+" "+String(UnitID_Part_3, HEX));
  Serial.println("RF24Network/examples/helloworld_rx/");
  
  pinMode(RFM_enbl, OUTPUT);
  digitalWrite(RFM_enbl, HIGH);
  pinMode(RED_led, OUTPUT_OPEN_DRAIN);
  pinMode(GRN_led, OUTPUT_OPEN_DRAIN);
  pinMode(BLU_led, OUTPUT_OPEN_DRAIN);
  pinMode(BAT_sens, INPUT);
  digitalWrite(RED_led, HIGH);
  digitalWrite(GRN_led, HIGH);
  digitalWrite(BLU_led, HIGH);
  
  SPI.begin();
  radio.begin();
  network.begin(/*channel*/ 90, /*node address*/ this_node);
}

void loop(void){
  
  network.update();                  // Check the network regularly

  
  while ( network.available() ) {     // Is there anything ready for us?
    
    RF24NetworkHeader header;        // If so, grab it and print it out
    payload_t payload;
    network.read(header,&payload,sizeof(payload));
    Serial.print("Received packet #");
    Serial.print(payload.counter);
    Serial.print(" at ");
    Serial.println(payload.ms);
    digitalWrite(GRN_led, LOW);
    delay(500);
    digitalWrite(GRN_led, HIGH);
  }
}`

TMRh20, 2020-09-17:
I would suggest testing with slightly modified examples or enable debugging to see what is going on for yourself: https://github.com/nRF24/RF24Network/blob/master/RF24Network_config.h#L56

diygoodies, 2020-09-18:
/** Debug Options */
    #define SERIAL_DEBUG
    //#define SERIAL_DEBUG_MINIMAL
    #define SERIAL_DEBUG_ROUTING
    //#define SERIAL_DEBUG_FRAGMENTATION
    //#define SERIAL_DEBUG_FRAGMENTATION_L2
    /*************************************/
Doesn't make any changes, any additional  debug info  doesn't appear

TMRh20, 2020-09-18:
Crap, the debugging output uses printf, I guess its not enabled on STM32.

You could try the available libraries like LibPrintf to enable it, but I'm not sure if it will work without additional modifications due to the use of program memory related functions. ( see https://github.com/nRF24/RF24/issues/414 and https://github.com/nRF24/RF24/blob/master/RF24_config.h#L155 )

2bndy5, 2020-09-18:
Damn, I just came across a blog post about getting this library working on the STM33F40x. The only problem was addressing the printf problem. It was an old thread (~2016). I'll check my browser history and report back

diygoodies, 2020-09-18:
Replaced pritf with set of serial.print. Here is log of #01 Child

Start 1
UID [HEX]  : 66aff55 56487850 87083317
RF24Network/examples/helloworld_rx/
Received packet #2517 at 3777000
Received packet #2520 at 3781500
Received packet #2523 at 3786000
Received packet #2526 at 3790500
Received packet #2529 at 3795000
Received packet #2532 at 3799500
Received packet #2535 at 3804000
Received packet #2538 at 3808500
39514MAC Send fail to 11 via 11 on pipe 5
Received packet #2541 at 3813000
Received packet #2544 at 3817500
Received packet #2547 at 3822000
Received packet #2550 at 3826500
Received packet #2553 at 3831000
Received packet #2556 at 3835500
Received packet #2559 at 3840000
Received packet #2562 at 3844500
Received packet #2565 at 3849000
Received packet #2568 at 3853500
Received packet #2571 at 3858000
Received packet #2574 at 3862500
Received packet #2577 at 3867000
Received packet #2580 at 3871500
Received packet #2583 at 3876000
107014MAC Send fail to 11 via 11 on pipe 5
Received packet #2586 at 3880500
Received packet #2589 at 3885000
117514MAC Send fail to 111 via 11 on pipe 5
Received packet #2592 at 3889500
Received packet #2595 at 3894000
126514MAC Send fail to 111 via 11 on pipe 5
Received packet #2598 at 3898500
129514MAC Send fail to 11 via 11 on pipe 5
Received packet #2601 at 3903000
Received packet #2604 at 3907500
140014MAC Send fail to 111 via 11 on pipe 5

diygoodies, 2020-09-18:
Here is log of #11 Child

RF24Network/examples/helloworld_rx/
Received packet #2344 at 3517500
7587MAC Send fail to 111 via 111 on pipe 5
Received packet #2347 at 3522000
12087MAC Send fail to 111 via 111 on pipe 5
Received packet #2350 at 3526500
16588MAC Send fail to 111 via 111 on pipe 5
Received packet #2353 at 3531000
21088MAC Send fail to 111 via 111 on pipe 5
Received packet #2356 at 3535500
25588MAC Send fail to 111 via 111 on pipe 5
Received packet #2359 at 3540000
30088MAC Send fail to 111 via 111 on pipe 5
Received packet #2362 at 3544500
34588MAC Send fail to 111 via 111 on pipe 5
Received packet #2365 at 3549000
39088MAC Send fail to 111 via 111 on pipe 5
Received packet #2368 at 3553500
43589MAC Send fail to 111 via 111 on pipe 5
Received packet #2371 at 3558000
48089MAC Send fail to 111 via 111 on pipe 5
Received packet #2374 at 3562500
52589MAC Send fail to 111 via 111 on pipe 5
Received packet #2377 at 3567000
57089MAC Send fail to 111 via 111 on pipe 5

diygoodies, 2020-09-20:
Ok I've made it work. Just swaped NRF modules between 011 an 0111 children. I have no clue what is the difference in that two modules. https://youtu.be/Sr2gImN_1ow

2bndy5, 2020-10-09:
@Avamander @diygoodies This issue is resolved. Probably a power issue related to either using batteries or using a breakout board for the nRF24L01+ (as seen in the video) -- ultimately it was a hardware problem. So, please close this issue.

closed 2020-10-09 by Avamander