nRF24/RF24Network #161: deploy documentation using Github Pages

2bndy5, 2020-11-28:
This issue is part of an effort to start hosting nRF24 org repos' docs on github per individual repo instead of uploading all RF24* lib's docs to tmrh20.github.io.

I've created a github workflow that builds the docs on every commit. This same workflow will upload the generated HTML files to the "gh-pages" branch upon publishing a new release. Notice I've already taken the liberty to create the "gh-pages" branch for this repo.

1. For this workflow to work properly, this repo's settings needs to have the "gh-pages" branch set to the Github Pages Source option:
[image]
    After enabling this option, I expect the first deployment to Github Pages to render an error message.

2. Additionally, it would less visually noisy if the "packages" & "environments" sections are hidden in the repo's sidebar settings:
[image]

When these stipulations have been met in the repo settings I will submit a PR

2bndy5, 2020-11-28:
I'm editing all doc links (not RF24 install script link for all RF24 libraries) to the nRF24 org repos. I found a broken link that leads to tmrh20.github.io/RF24Network_Dev. Should it be without the "_Dev" part?

TMRh20, 2020-11-29:
Yeah, all the network_dev links should be changed to just network.

closed 2020-11-29 by TMRh20