nRF24/RF24Network #165: Problem with Arduino NANO 33 IoT

piotr199809, 2021-01-13:
Hello! I want to upload a program using the RF24Network library to Arduino NANO 33 IoT board, but I get errors during compilation.
Errors:

Arduino:1.8.13 (Windows 10), Płytka:"Arduino NANO 33 IoT"

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:28:6: error: 'Sync' has not been declared

 void Sync::update(void)

      ^~~~

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp: In function 'void update()':

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:31:3: error: 'network' was not declared in this scope

   network.update();

   ^~~~~~~

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:37:16: error: 'len' was not declared in this scope

   while ( at < len )

                ^~~

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:39:10: error: 'app_data' was not declared in this scope

     if ( app_data && internal_data && app_data[at] != internal_data[at] )

          ^~~~~~~~

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:39:10: note: suggested alternative: 'update'

     if ( app_data && internal_data && app_data[at] != internal_data[at] )

          ^~~~~~~~

          update

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:39:22: error: 'internal_data' was not declared in this scope

     if ( app_data && internal_data && app_data[at] != internal_data[at] )

                      ^~~~~~~~~~~~~

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:59:42: error: 'to_node' was not declared in this scope

     RF24NetworkHeader header(/*to node*/ to_node, /*type*/ 'S' /*Sync*/);

                                          ^~~~~~~

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:59:42: note: suggested alternative: 'tone'

     RF24NetworkHeader header(/*to node*/ to_node, /*type*/ 'S' /*Sync*/);

                                          ^~~~~~~

                                          tone

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:90:9: error: 'app_data' was not declared in this scope

         app_data[pos] = val;

         ^~~~~~~~

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:90:9: note: suggested alternative: 'update'

         app_data[pos] = val;

         ^~~~~~~~

         update

C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp:91:9: error: 'internal_data' was not declared in this scope

         internal_data[pos] = val;

         ^~~~~~~~~~~~~

exit status 1

Błąd kompilacji dla płytki Arduino NANO 33 IoT.

What is wrong? How to fix it?

2bndy5, 2021-01-13:
+1 I was able to replicate this using the arduino library manager to install the RF24 & RF24Network libs.

TMRh20, 2021-01-14:
You can probably just delete the C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp and Sync.h files. This should probably be done for the next release as these files are no longer used in the library.

> On Jan 13, 2021, at 5:14 PM, piotr199809 <notifications@github.com> wrote:
> 
> C:\Users\piotr\Documents\Arduino\libraries\RF24Network\Sync.cpp

2bndy5, 2021-01-14:
yep, that fixed the compilation errors. @TMRh20 I can add an Arduino CI workflow like I did for the RF24 lib. The Arduino Nano 33 IoT board is included already, and a library (like RF24) can be specified as a dependency.

TMRh20, 2021-01-14:
@2bndy5 Sure, that'd be great

piotr199809, 2021-01-14:
Thanks! I removed Sync.cpp and Sync.h and it helped

PR #166 (2021-01-15): Add Arduino CI
This adds a CI workflow to compile the examples against AVR and SAMD cores. Also I formatted the examples using AStyle formater (as used in the Arduino IDE). Additionally, I replaced the direct calls to `printf()` with `Serial.print*()` (its likely I missed some) in the examples. This also closes #165 by removing the _sync.*_ files per recommendation of @TMRh20 

### Exceptions
I had to skip checking a couple examples:
1. _Network_Ping_ 
    - The Arduino CLI doesn't seem to have "avr/pgmspace.h" included
2. _Network_Ping_Sleep_
    - had trouble with this one because the `RF24N [... cut]
closed 2021-01-15 by TMRh20