nRF24/RF24Network #167: Connecting an NRF24L01 module to a 2nd SPI on a samd21g18

ex-caliper, 2021-02-09:
Hi,  

Having previously created a small network of NRF24L01 modules with Arduino Uno’s, I now have a need increase capacity in the master device.  Consequently,  I now have a custom breakout board with a samd21g18 containing an Arduino Zero bootloader. 

With the default SPI occupied, I am trying to connect the NRF24L01 module to a 2nd SPI and like most I have studied the Adafruit and Sparkfun tutorials on how to create a 2nd SPI.  https://learn.sparkfun.com/tutorials/adding-more-sercom-ports-for-samd-boards/introduction

Hence, I created an SPIClass mySPI (&sercom1, 11, 12, 10, SPI_PAD_2_SCK_3, SERCOM_RX_PAD_0); 

However, where I am lost is how to communicate with the NRF24L01 device on this 2nd SPI.  I cannot even get the device to work on the default SPI.  

How do I initialise the device, is this done by using the mySPI.beginTransaction(SPISettings(10000000, MSBFIRST, SPI_MODE0)) and further commands entered through SPI.transfer? 

Has anyone successfully connected the module to a samd21g18, either by the default SPI or more interestingly via a 2nd SPI?

Any information would be greatly appreciated, thanks in advance.

2bndy5, 2021-02-09:
Hope this helps:
- SPI bus is initialized in [`RF24::begin()`](https://github.com/nRF24/RF24/blob/0281ea2048f5295d382257b51980f0004a0f4d1f/RF24.cpp#L665)
- All SPI transactions are preceded by [`RF24::beginTransaction()` (private method)](https://github.com/nRF24/RF24/blob/0281ea2048f5295d382257b51980f0004a0f4d1f/RF24.cpp#L86)
- someone else had a [similar question about the ESP32](https://github.com/nRF24/RF24/issues/722).

I think this is another reason to let `begin()` take an SPI object reference.

2bndy5, 2021-02-09:
I have it correctly working on a QtPy (Seeed Xiao clone) that's running on a ATSAMD21E18. I also had success on a Adafruit Feather running on a ATSAMD21G18. Just don't expect `radio.printDetails()` to work because of something the Arduino SAMD core does for `printf()` (that's [a separate issue](https://github.com/nRF24/RF24/issues/414)).

ex-caliper, 2021-02-10:
Thankful to @szerwi for his reply, however this seems to be only applicable to ESP32's SPI.cpp  https://github.com/nRF24/RF24/issues/722  and unfortunately not skilled enough to cross reference to samd21g18.
@2bndy5, thank you for previous information. You say you had success with Adafruit Feather running on a ATSAMD21G18, did you use the SPI Legacy header, i.e. MOSI(PB10), MISO(PA12) and SCK(PB11) CE & CS to any.  Also did you have to insert all of those in the RF24 radio constructor, whereas normally with a default SPI, only CE and CS are required. RF24 radio(CE, CS) 
Still having difficulty getting the module to work even with the default SPI, having parked the idea of getting it to run on a 2nd SPI at the moment and getting back to basics.  Is there a simple sketch you could point me to please.

2bndy5, 2021-02-10:
>did you use the SPI Legacy header, i.e. MOSI(PB10), MISO(PA12) and SCK(PB11) CE & CS to any

Nope. Not sure what that even is off-hand

> Also did you have to insert all of those in the RF24 radio constructor, whereas normally with a default SPI, only CE and CS are required. RF24 radio(CE, CS)

Nope. You might beware what SparkFun says the library constructor can/cannot take as parameters. Check [Lib docs if not sure](https://nrf24.github.io/RF24/classRF24.html#abe8f878e238644252ccf5ec27fba611d).

TBH, I started developing the new examples on the Adafruit Feather M0 Express ("Express" means that there's extra storage memory on a different SPI bus), so the examples' code is all I need to get the library working on the default spi bus (I remember having to change the CE & CSN pins to `RF24 radio(5, 6)`).

Changing to a secondary SPI bus when a board isn't using the Linux SPIDEV driver is a more popular problem lately.

ex-caliper, 2021-02-13:
@2bndy5 With regard to SPI Legacy header, apologies rubbish diagram, this is the ICSP.
After dismantling the breadboard and re-wiring it, I can now confirm with you that running an nRF24L01 module with an MSP2807(ILI9341) tft on a samd21g18 default SPI will work. I have no idea why it didn't work in the first place.
For those with a simple breakout board: MOSI(PB10), MISO(PA12), SCK(PB11) Note: MISO not required to tft.
In my case, made the DC, CS and RESET pins of the tft, and CE, CSN pins of the nRF24L01 high in setup. e.g digitalWrite(CS, HIGH); to avoid contention. 
2nd SPI on samd21g18, still work in progress..

2bndy5, 2021-02-13:
Oh I didn't know you were working with a TFT as well (sounds like a ST7789). You should be able to use both devices on the same bus as long as your TFT has a CS pin.

2bndy5, 2021-03-20:
@ex-caliper I'm working on this feature... I think its ready for testing (if you want to try out). The [RF24 repo's overload-begin branch](https://github.com/nRF24/RF24/tree/overload-begin) has the changes. Thanks to your tip about the Sparkfun tutorial, I've added [example source code specific for the ATSAMD21](https://github.com/nRF24/RF24/blob/overload-begin/docs/arduino.md#atsamd21-example)