nRF24/RF24Network #168: what's the utility of addressOfPipe()?

2bndy5, 2021-03-07:
I see `addressOfPipe()` declared and defined, but I can't find any source code reference that actually calls `addressOfPipe()`. It seems like a publicly exposed version of `pipe_address()`, but for logical addresses (not the physical addresses - thus no consideration for multicasting). Is this function more significant in RF24Mesh?

Please excuse my seemingly naive confusion; I'm still trying to figure out the mechanics of this lib (haven't familiarized myself with RF24Mesh yet). Although, I feel like I'm starting to understand enough to ask this (guess what I'm porting to pure python 😉).

TMRh20, 2021-03-07:
I'm really not sure where this function came from other than experimentation. I also fail to find anywhere where it is actually used, so it can probably be removed. 

I confuse myself sometimes looking back at old code. :p

2bndy5, 2021-03-07:
I'm skeptic about removing something that's been out there for so long. Maybe just mark it for deprecation.

TBH, your answer is both surprising and reassuring; I think I've finally got a good grasp on this library.

ps - I took the time to fix the formatting on my fork (inconsistent indentation made it difficult to read). I think the python examples are a bit out of date, so you might see a PR from me.

Avamander, 2021-03-07:
> Maybe just mark it for deprecation.

If it's the only public API for accessing that data specifically, maybe there's no need.

2bndy5, 2021-03-07:
@Avamander What do mean by "there's no need"? I'm not against just closing this issue and leaving it be; I'm just trying to understand things better.

Avamander, 2021-03-07:
> @Avamander What do mean by "there's no need"? I'm not against just closing this issue and leaving it be; I'm just trying to understand things better.

You already expressed concern about backwards compatibility when removing it, I was just wondering why even mark it as deprecated (unless there's a practical need?).

2bndy5, 2021-03-08:
Good point. It seems there could be a case made for either practicality or impracticality. Better off just leaving it as is and letting the user decide weather to use it or not.

closed 2021-03-08 by 2bndy5

2bndy5, 2021-07-28:
I still can't figure out what this function is meant to provide. As far as I can tell it just shifts the pipe number into any given number based on the network level of the node that this function is called on.
```python
Network node address__ 0o11
>>> oct(nrf.address_of_pipe(0o13, 3))
'0o33'
>>> oct(nrf.address_of_pipe(0o11, 3))
'0o31'
>>> oct(nrf.address_of_pipe(0o11, 4))
'0o51'
>>> oct(nrf.address_of_pipe(0o11, 2))
'0o31'
>>> oct(nrf.address_of_pipe(0o1, 2))
'0o21'
>>> oct(nrf.address_of_pipe(0o211, 3))
'0o231'
>>> oct(nrf.address_of_pipe(0o5040, 5))
'0o5050'
```

Leaving this one out of my circuitpython API to save memory.

TMRh20, 2021-07-28:
There seems to be no purpose for it. I can’t remember why its there.