nRF24/RF24Network #172: Missing byte in address_translation array.

adrmcintyre, 2021-04-12:
https://github.com/nRF24/RF24Network/blob/ecb31c222bb9499a191cb2a303a82f4e2b13c02e/RF24Network.cpp#L1155

Looks like there's a missing byte ```0xcc```. It would be wise to initialise with an explicit array size as follows instead:

```C++
static uint8_t address_translation[8] = { 0xcc,0xc3,0x3c,0x33,0xce,0x3e,0xe3,0xec };
```

2bndy5, 2021-04-12:
The [`0xCC` byte is used](https://github.com/nRF24/RF24Network/blob/ecb31c222bb9499a191cb2a303a82f4e2b13c02e/RF24Network.cpp#L1156) for all bytes in the `result` buffer before continuing with translation. 

>It would be wise to initialise with an explicit array size

Agreed, but the compiler (at least for the Arduino IDE) outputs an error for explicitly sized arrays when declaring an array with initialized values.

2bndy5, 2021-04-12:
@adrmcintyre I see you forked ManiacBug's RF24 repo, but I have to warn you that repo is no longer maintained. we are currently maintaining the RF24 lib [here](http://github.com/nRF24/RF24) at the nRF24 org

adrmcintyre, 2021-04-12:
I was concerned that the subsequent ```% 8``` returns a value from 0-7, and that 7 would index outside of this array, but actually I see ```is_valid_address``` guarantees the octal digits will not include 7, so this function is safe as written. Nevertheless this is not clear at first glance.

Thanks for the tip about the fork which I've now deleted. I'll be sure to stick to the supported version

closed 2021-04-12 by adrmcintyre

2bndy5, 2021-04-12:
I recently did a deep dive into the RF24Network source code, and practically none of it seemed clear at first glance. I now understand that much the confusing parts were accommodations for RF24Mesh.