nRF24/RF24Network #177: get user input in examples?

2bndy5, 2021-06-19:
I've returned to porting this lib into my circuitpython lib... I noticed the examples here have hard-coded node addresses. Is there a reason for this (proper addressing demo)? 

Initially, I was thinking ask the user for an octal number at program start-up to more dynamically set the node address.

As for the node targeted for outbound messages, let
- any non-master node transmit to master 
- any master node transmit to node 01 (or maybe ask the user for this address also)
 
I suppose it would be prudent to ping the parent node when the user enters the node address to be used, but I'm not that knowledgeable about RF24Mesh yet. I think the ping type messages are meant for RF24Mesh but not exclusively as the code is integrated in RF24Network.
<br>
ps - I have other ideas to clarify the examples' output like not using the payloads' `millis()` value as the receiving node's timestamp (instead just declare the timestamp as part of the received message).

TMRh20, 2021-06-19:
No reason for hard coded addresses other than simplicity. 
The network ping capability should work fine, but may confuse users as it is a system message not a user message.

2bndy5, 2021-06-19:
> The network ping capability should work fine, but ...

Good point. On second thought, we would need to instantiate the network object using the node address before we ping the parent node. This would be a counter-productive check since we can't re-adjust the node address after instantiating it. It would be more informative to prompt the user to "make sure the parent node is running" using the returned address from `RF24Network::parent()`.

closed 2021-06-19 by 2bndy5