nRF24/RF24Network #178: Leonardo freeze with Sending and Receiving

SShattered, 2021-06-26:
I can send data from **Teensy** 3.2 to Arduino Leonardo (In XInput) to control right joystick, so just one way communication. But I want to send feedback of joystick to **Teensy**. However, after sending data from Leonardo to Teensy, it gets freeze and I can't no longer control joystick with my Teensy which indicates data not received by leonardo. This happens only after I send data from the Leonardo to Teensy, I receive few data which I print on Teensy Serial Monitor, then after few seconds It stops receiving and Leonardo seems to be stopped.
 Below is the code for Leonardo,

```
#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>
#include <XInput.h>

#define MAX_PAYLOAD 64

RF24 radio(8, 10);         

RF24Network network(radio);      
const uint16_t this_node = 00;
const uint16_t left_node = 01;

int headAccelX = 0;
int headAccelY = 0;

char headBuffer[MAX_PAYLOAD]; //Head

uint8_t rumbleValue = 125;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  XInput.setAutoSend(false);
  XInput.setJoystickRange(-100, 100);
  XInput.begin();
  
  SPI.begin();
  radio.begin();  
  radio.setDataRate(RF24_2MBPS);
  radio.setPALevel(RF24_PA_MAX);
  network.begin(95, this_node);

  delay(500);
}

void parse() {
  XInput.setJoystick(JOY_RIGHT, headAccelX, headAccelY);
  XInput.send();
}

void loop() {
  //Test
  char rumble[MAX_PAYLOAD];
  sprintf(rumble, "%i", rumbleValue);
  RF24NetworkHeader headers(left_node, 'V');
  network.write(headers, &rumble, MAX_PAYLOAD);
  
  network.update();                  
  while (network.available()) { 
    RF24NetworkHeader header;
    int dataSize = network.peek(header);
    switch(header.type){
      case 'H':
        network.read(header, &headBuffer, dataSize);
        sscanf(headBuffer, "%i;%i",
                    &headAccelX,
                    &headAccelY);          
      break;
      default:
        network.read(header, 0, 0);
      break;
    }
  }
}
```

In teensy I have added code for receiving data, it does not give any problem. Problem arise when I try to send data from Leonardo to Teensy. 

Teensy Code,

```
void sendData()
{
  char buffer[MAX_PAYLOAD];
  sprintf(buffer, "%i;%",
          accelX,
          accelY);
  RF24NetworkHeader header(base_node, 'L');
  network.write(header, &buffer, MAX_PAYLOAD);

  //Test
  network.update();                  
  while (network.available()) {
    char buf[16];
    RF24NetworkHeader headers;
    int dataSize = network.peek(headers);
    switch(headers.type)
    {
      case 'V':
        network.read(headers, &buf, dataSize);
        Serial.println(buf);
      break;
      default:
        network.read(headers, 0, 0);
      break;  
    }
  }
}
```

Teensy doesn't get freeze in any situation, I had tried printing values on serial monitor.

2bndy5, 2021-06-26:
I think you have a typo. 

In your Leonardo code
>  ```cpp
> sscanf(headBuffer, "%i;%i", &headAccelX, &headAccelY);
> ```

In your teensy code
> ```cpp
> sprintf(buffer, "%i;%", accelX, accelY);
> ```

You're missing the second `i` in the teensy code. I don't know if this will fix your problem though.

SShattered, 2021-06-27:
> I think you have a typo.
> 
> In your Leonardo code
> 
> > ```c++
> > sscanf(headBuffer, "%i;%i", &headAccelX, &headAccelY);
> > ```
> 
> In your teensy code
> 
> > ```c++
> > sprintf(buffer, "%i;%", accelX, accelY);
> > ```
> 
> You're missing the second `i` in the teensy code. I don't know if this will fix your problem though.
> 
> Also in your teensy code:
> 
> > ```c++
> > char buf[16];
> > ```
> 
> This buffer size seems a little small to me. Try using the return value of `peek(header)`
> 
> ```c++
> const int dataSize = network.peek(headers) + 8;
> char buf[dataSize];
> ```

Sorry for that. I was reducing the data and missed "i". Behavior is still same, still freezes the Leonardo. At first I was using the buffer size of 64. It didn't change anything either :(

2bndy5, 2021-06-27:
Damn, I was hoping it would be that simple. I do find it weird that you're using c-strings as data buffers (which require an extra null terminating byte) instead of a data `struct`. In my experience, a frozen MCU (with native USB support) usually means I misused a pointer or malformed a c-string (aka "segmentation fault").

Without seeing your code in it's entirety, I'd ask you to first test the library examples to make sure it's not the library code and that the boards are compatible with the library (the Xinput bootloader on the Leonardo might compromise that).

SShattered, 2021-06-27:
> Damn, I was hoping it would be that simple. I do find it weird that you're using c-strings as data buffers (which require an extra null terminating byte) instead of a data `struct`. In my [...]

Thanks! I will test it out and let you know. I will see using struct

closed 2023-05-17 by TMRh20