nRF24/RF24Network #179: [src clarity] use macros for some sentinels

2bndy5, 2021-06-28:
Ok, I think I'm finally understanding the `write()` functions' `directTo` & `writeDirect` (poorly named BTW) parameters. I have noticed that there are sentinel values used in places where a noob would expect to see a valid octal address value. I would like permission to add 2 new macros whose names would clarify why/where the sentinel values are used.
1. `#define NETWORK_AUTO_ROUTING 070` This would be used in place of the `directTo` parameter's default value (passed from the overloaded `write()`) as well as in the conditional checks like `if (directTo != 070)`.
   - This macro doesn't seem to be needed in RF24Mesh
2. `#define NETWORK_MULTICAST_ADDR 0100` This would be used where the `header.to_node` is set to 0100 for multicasted messages. We could also use this macro in conditional checks like `if (header.to_node == 0100)`.
   - The use of this macro would actually align with the existing `NETWORK_DEFAULT_ADDR`.
   - This macro could also be used in [`RF24Mesh::requestAddress()` on this line](https://github.com/nRF24/RF24Mesh/blob/a343ac4210f8351a722703b1dd508e534bd1f26c/RF24Mesh.cpp#L288)

I doubt that using these new macros will have any kind of performance/memory impact.

I've already taken the liberty of adjusting some of the calls to `write()` where a macro's value is passed as the `directTo` parameter instead of the macro name (see commit https://github.com/nRF24/RF24Network/commit/251255e7d32fba40dca8b41fcf5151198498ca99).

Additionally, I'd _really_ like to rename the `writeDirect` parameter to something that doesn't sound like a duplicate of `directTo`. My suggestions are `sendType`/`writeType` or `sendBehavior`/`writeBehavior`. I went with `send_type` in my python porting of this lib, but that can/might be changed before rolling out RF24Network in my CirPy lib. This renaming would better align with the behavior of the `RF24::startFastWrite()` function's `multicast` parameter because it changes how the payload is transmitted and not where the payload is transmitted to.

TMRh20, 2021-06-28:
> I've already taken the liberty of adjusting some of the calls to write() where a macro's value is passed as the directTo parameter instead of the macro name (see commit 251255e).

Sounds good.

2bndy5, 2021-06-29:
> Additionally, I'd _really_ like to rename the `writeDirect` parameter to something that doesn't sound like a duplicate of `directTo`. My suggestions are `sendType`/`writeType` or `sendBehavior`/`writeBehavior`. 

Oops! I got it backwards (yet again): The `writeDirect` param is adequately named; its the `directTo` param that needs clarity. I'm going with `sendType` instead of `directTo` because `sendType` is the scoped variable name that helped clear up the confusion for me (see [here](https://github.com/nRF24/RF24Network/blob/d8bcb226336652b5409c2689b3922a6d8504bee9/RF24Network.cpp#L787-L797)).

2bndy5, 2021-07-23:
Found a comment that seems like an artifact from early development

https://github.com/nRF24/RF24Network/blob/d8bcb226336652b5409c2689b3922a6d8504bee9/RF24Network.cpp#L562

I'm guessing this has been tested since the lib is so seasoned by now. I'll just remove it to abate any confusion.

2bndy5, 2021-07-23:
### concerning ATTiny defaults
It would seem that `MAX_PAYLOAD_SIZE` is only used if the user comments out `#define DISABLE_FRAGMENTATION` in _RF24Network_config.h_. Using the current defaults, the actual max message size transmitted is 24. I'll add a comment to the `#define MAX_PAYLOAD_SIZE 72` line about only being applicable when fragmentation is enabled. Furthermore, I think there are some comments in the same code block (lines 72-73) trying to imply this point:

https://github.com/nRF24/RF24Network/blob/d8bcb226336652b5409c2689b3922a6d8504bee9/RF24Network_config.h#L65-L76

However, it is not mentioned that only using the radio's 3 levels in the TX FIFO is recommended for ATTiny devices.

closed 2021-11-14 by TMRh20