nRF24/RF24Network #180: [question] is message size actually transmitted with payload?

2bndy5, 2021-07-02:
### Situation
This has been bugging me because the docs say
>Frames: Header (8-bytes) + Frame_Size (2-bytes) + Data (?-bytes) 

and I do see a `uint16_t message_size` member in `RF24NetworkFrame` definition...

I'm currently able to accurately receive & disassemble the network payloads from the Arduino examples in my cirPy lib and I don't see the message size in the payload. 
### Question
Could the message size be transmitted conditionally (like only with certain message types)? Is the message size determined dynamically from the sum of payload size(s) fetched from radio RX FIFO (- 8 for each header)? Could the docs be inaccurate?

TMRh20, 2021-07-02:
The header.reserved is what holds the value for number of fragments which is set for fragmented payloads. Otherwise size is determined by the receiving radio.

2bndy5, 2021-07-02:
Ok so the docs must be talking about the memory allocated for the `frame_buf`. I understood the `reserved` member (that has very good docs).

Sorry if this seemed like a dumb question. I figured I'd inject as much clarity in the network docs as I could while porting it all to python.

closed 2021-07-02 by 2bndy5