nRF24/RF24Network #181: [Request] make maxlen parameter to `read()` & `peek()` optional

2bndy5, 2021-07-23:
After helping resolve nRF24/RF24Mesh#184, I found that making the user specify (every time) the length of message to fetch is a bit unintuitive because the frame has the information (`message_size` member) to use as a default value. To ensure backward compatibility, we could use the `MAX_PAYLOAD_SIZE` as the maxlen parameter's default value, then use `rf24_min()` to limit the length of the message read to the message's max size. 

Note: The `read()` function already does use `rf24_min()`, but the `peek(&header)` does not (though it should anyway).

2bndy5, 2021-07-23:
Is there a reason that `read()` doesn't use `if (!available()) { return 0; }` to prevent trying to read from an empty queue?

TMRh20, 2021-07-23:
Not really, it assumes the user has previously called the available(); function.

2bndy5, 2021-07-23:
Instead of adding the extra instructions to prevent fetching data from an empty queue, I will add comments and a `@note` to the `read()` function's docs.

closed 2021-11-14 by TMRh20