nRF24/RF24Network #183: [Request] make `level` parameter to multicast() optional

2bndy5, 2021-07-28:
I had the thought to do this in when porting this lib to pure python. If the `level` parameter is not specified, then we can let the `multicast()` function automatically use the private member `multicast_level` (set by `setup_address()` or `multicastLevel()`) as a default value. This should be helpful because the user has no way of getting the current multicast level in the public API (or maybe I missed something).

It's my understanding that the `multicastLevel()` sets the network level in which the node is listening for multicasted messages. Making the `level` parameter optional, will relieve the user of having to keep passing the new multicast level to the `multicast()` function.

ps - I'll be conditionally adding the `multicastLevel()`, `multicast()`, and `multicastRelay` to the python wrapper for completeness. Additionally, I'm also adding some other missing attributes like `networkFlags` & `routeTimeout` (I've already added the missing `begin(uint16_t)` method).

TMRh20, 2021-07-29:
Sounds good to me

2bndy5, 2021-07-29:
I've decided to fill out the missing docs for private members while everything is still fresh in my head...

In the docs about `multicastLevel()`, it says that for the level parameter:
>Levels 1 to 6 are available

Is this a typo? I thought there could only be 4 network levels (0-3) judging from the behavior of `RF24Mesh::requestAddress()` and the [topology page's image](https://github.com/nRF24/RF24Network/blob/master/images/topologyImage.jpg).

The topology page on network levels is kinda vague because it constantly compares to typical IP addressing scheme (which I take for granted/don't know). For my CirPy docs, I have [generated some graphs (using graphviz lib)](https://circuitpython-nrf24l01.readthedocs.io/en/rf24-network/network_docs/topology.html#network-levels) instead of manually using GIMP, but now I'm wondering if I conceived network levels wrong.

2bndy5, 2021-07-29:
This solution is testing well.

any objections to letting `multicastLevel()` return the current configured multicast level.
say like so
```py
>>>network.multicastLevel()
0
>>>network.multicastLevel(1)
1
```
This would at least allow mesh users to adjust it programmatically.

TMRh20, 2021-07-29:
I think that must be a typo but there should be 5 multicast levels, 0-4. Nodes n111-n555 make up level 4

2bndy5, 2021-07-29:
Thanks for the clarification. That was the last point of confusion for me. I'll fix the typo and leave the `multicastLevel()` function as is (I was overthinking it again).

closed 2021-11-14 by TMRh20