nRF24/RF24Network #185: Are some networkFlags deprecated?

2bndy5, 2021-08-05:
I can't find any reference in any of the RF24* libs using the `FLAG_BYPASS_HOLDS` & `FLAG_HOLD_INCOMING` values.

I blindly ported the `networkFlags` to pure python, but I'm only using the `FLAG_NO_POLL` in my RF24Mesh class. 😆

TMRh20, 2021-08-05:
Those should probably be removed, from some experimental code.

2bndy5, 2021-08-05:
no objections to that. It would actually grant any external system built on RF24Network more room for custom flags.

2bndy5, 2021-08-05:
It will look kinda weird in the docs that only bit positions 2 & 3 are used in the `networkFlags` member, but I've grown unusually comfortable with awkwardness like that.

TMRh20, 2021-08-06:
Hahaha, yeah I’m a messy coder. :/

2bndy5, 2021-08-06:
I have worked with much worse...

commit 17944505 (2021-08-05): resolves #185
closed 2021-11-14 by TMRh20