nRF24/RF24Network #186: RF24NetworkFrame.toString() has no definition

2bndy5, 2021-08-11:
I think I found another uncompleted relic from development. The _RF24Network.h_ declares a [`toString()` function for `RF24NetworkFrame`](https://github.com/nRF24/RF24Network/blob/d8bcb226336652b5409c2689b3922a6d8504bee9/RF24Network.h#L314-L323), but there is no definition for it in _RF24Network.cpp_. There aren't any references to `RF24NetworkFrame::toString()` in any debug statements (including the ones that are commented out).

I don't think there is a need to have `toString()` defined for `RF24NetworkFrame` objects. Furthermore, this is a function that the user could write themselves (in conjunction with `RF24NetworkHeader::toString()`) if their application wants/needs to. Similar functionality is [exhibited in the debug statements](https://github.com/nRF24/RF24Network/blob/d8bcb226336652b5409c2689b3922a6d8504bee9/RF24Network.cpp#L151).

ps - `RF24NetworkHeader::toString()` is much more convenient/applicable than having a `RF24NetworkFrame::toString()`.

TMRh20, 2021-08-11:
I agree, no need for frame.toString

2bndy5, 2021-08-12:
ok I'll remove the `RF24NetworkFrame::toString()` declaration. This will also remove the accompanied docs that doxygen is currently outputting.

closed 2021-11-14 by TMRh20