nRF24/RF24Network #187: [Request] make multicast_level and node_address members protected

2bndy5, 2021-08-11:
This feature request is similar to nRF24/RF24#782 in that I'll be using inheritence to provide additional functionality in the newer python wrappers that isn't available in the the C++ libs (for low-memory reasons).

With this feature added, I do plan on letting users set the `multicast_level` and `node_address` members via existing functionality (`RF24Network::multicast_level()` and `RF24Network::begin()` respectively). The following psuedo-code might help to better describe my intentions here
```py
import pyRF24.rf24 as rf24
import pyRF24.rf24_network as rf24_network

THIS_NODE = 0o1

radio = rf24.RF24(22, 0)
if not radio.begin:
    print("radio hardware not responding")
    exit()

network = rf24_network.RF24Network(radio)

# new python wrapper's setter for the node's logical address
# keep in mind that channel should be set with radio.set_channel(new_chnl) or radio.channel = new_chnl
network.node_address = THIS_NODE  # underlying functionality calls RF24Network::begin(THIS_NODE)
# new python wrapper's getter for the node's logical address
addr = network.node_address  # simply returns the private/protected member RF24Network::node_address

# set the multicast level
network.multicast_level = 1  # underlying functionality calls RF24Network::multicast_level()
# get the current multicast level
lvl = network.multicast_level  # simply returns the private/protected member RF24Network::multicast_level
```

TMRh20, 2021-08-11:
Yeah I think I’m good with this

2bndy5, 2021-08-11:
I'll also prefix the private/protected  `multicast_level` with a `_`. This is to not confuse with the public function by the same name.

closed 2021-11-14 by TMRh20