nRF24/RF24Network #190: Only transferring 24bytes of data

BryanInglis, 2021-12-09:
ESP32 <> Mega, trying to send a 32byte message from Esp32 > Arduino Mega and then a return 32byte message to the ESP32. I have debug options on and they show that its only sending a 24byte message. I've confirmed this by displaying the bytes received on both sides. I've checked the network config file and its not set for 24bytes. Am I missing something else or is their some other config that needs to be adjusted? The max msg size was set at 144 in the config. I'm using NRF24 + network library. I hit this late last night, so I'll update the library again tonight, just to make sure.

TMRh20, 2021-12-09:
Can you post a code example? Remember esp32 is 32-bit and mega is 8-bit so datatypes/sizes are the most likely issue. Ie: byte can take up 4 bytes of memory space on esp but only one byte on mega. See https://tmrh20.blogspot.com/2020/09/transferring-data-between-systems-using.html?m=1

BryanInglis, 2021-12-09:
Panic over, downloaded current version and issues solved. 2bndy5, no, checked that too. TBH, not sure why it stopped working, but the download sorted it right out with the default settings, so project back on track. Just to confirm the child nodes still respond correctly. Thanks to everyone who dropped a suggestion, may thanks. Bryan.

closed 2021-12-09 by BryanInglis