nRF24/RF24Network #192: Multicast on Python (Master) with Arduino-Nodes

SilentM4X, 2022-05-19:
I use a Raspberry Pi to communicate with 10 nodes via RF24Network. 
So far I've always addressed each node individually, which works quite well so far - but if devices are offline, there are delays and the code is very long:

            payload = pack('<LL', millis(), 0 )    
            network.write(RF24NetworkHeader(node2), payload)
            network.write(RF24NetworkHeader(node3), payload)
            network.write(RF24NetworkHeader(node4), payload)
            network.write(RF24NetworkHeader(node5), payload)
            network.write(RF24NetworkHeader(node012), payload)
            network.write(RF24NetworkHeader(node032), payload)
            network.write(RF24NetworkHeader(node042), payload)
            network.write(RF24NetworkHeader(node052), payload)
            network.write(RF24NetworkHeader(node053), payload)
            network.write(RF24NetworkHeader(node054), payload)
            network.write(RF24NetworkHeader(node055), payload)
            network.write(RF24NetworkHeader(node014), payload)
            network.write(RF24NetworkHeader(node015), payload)
            network.write(RF24NetworkHeader(node025), payload)
            network.write(RF24NetworkHeader(node035), payload)

Now I've read that there is network.**multicast** function, but I can only find information about handling in Arduino. I couldn't find informations how to use multicast on the Raspberry Pi with Python. 

Does anyone have an idea how to do that?

2bndy5, 2022-05-19:
Multicast support in the python wrapper is something that was overlooked/incomplete for long time. I added some it into the python wrapper for the last release.

>I can only find information about handling in Arduino

I'm curious what info source(s) you imply here. The [`RF24Network::multicast()` docs](https://nrf24.github.io/RF24Network/classRF24Network.html#a96a66907061c54ee1fea6ea7f78d6827) are meant to apply to both Linux and Arduino platforms.

> Does anyone have an idea how to do that?

This is rather broad question. Multicasting is dependent on understanding the [RF24Network topology](https://nrf24.github.io/RF24Network/md_docs_tuning.html) (there's a section on that page about multicasting). Ultimately, if there are inactive nodes, then it severely limits the capability of the network.

IIRC, the radios' pipe 0 for each level are set up to listen to for any frames (multicasted or not) aimed at them. So, if a normal `write()` fails, then you could try multicasting to a specific level (and handle the payload appropriately). However, in the absence of forwarding nodes, the origin/transmitting node must be in range of a node on the specified network level.

>but if devices are offline, there are delays and the code is very long

The [`RF24Network::txTimeout`](https://nrf24.github.io/RF24Network/classRF24Network.html#a8838b42717e7c9da37213b020db1f88c) and [`RF24Network::routeTimeout`](https://nrf24.github.io/RF24Network/classRF24Network.html#a657fd03730b6c9084cc440b4fe0177c3) can be used to reduce the delay. I would **_not_** recommend playing with [`RF24::setRetries()`](https://nrf24.github.io/RF24/classRF24.html#a4c6d3959c8320e64568395f4ef507aef) as it is used based on recommendations from the radio's datasheet and the context of the RF24Network operation.

SilentM4X, 2022-05-20:
Since I always get the following error, I assumed that the documentation might exclude Linux.

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "Server.py", line 529, in bums
    callotherButtons(oct(header.from_node))
  File "Server.py", line 115, in callotherButtons
    network.multicast(RF24NetworkHeader(node023), payload, sizeof(payload), 01)
AttributeError: 'RF24Network' object has no attribute 'multicast'
```

The fact that not all nodes are always connected (does not happen regularly), is that this projekt is a quiz with wireless buzzers and not everything is used every time (eg. lightbars, movingheads etc.).

It results that usually only the first level and a few of the second level are used. I already switched all my led spots to wireless DMX 2.4 GHz to reduce the trouble.

Since now, i never used [RF24Network::txTimeout] and [RF24Network::routeTimeout], the [RF24::setRetries()] are set to 
`radio.setRetries(10,15)` (with default values it won't work)

2bndy5, 2022-05-20:
>Since I always get the following error, I assumed ...

Please don't assume the devs know the entire context of your situation. I say this because it looks like this is the first issue you ever opened.

---

I think your setup is outdated. Python 2.7 is officially dead (since 2019 I believe) - the fact that your OS still has it tells me you're using an old RPi OS image. Furthermore, the [python wrapper does have the multicast function](https://github.com/nRF24/RF24Network/blame/ffcdf8d9bbdb0e78163ccb8d09e4b3d559e5976c/RPi/pyRF24Network/pyRF24Network.cpp#L141), but you need to update your version of RF24Network (to v1.0.16) as this function was added in #176.

> `radio.setRetries(10,15)` (with default values it won't work)

Are you calling the `RF24::setRetries()` after `RF24Network::begin()`? I ask because [this lib appropriates that setting during `begin()`](https://github.com/nRF24/RF24Network/blob/ffcdf8d9bbdb0e78163ccb8d09e4b3d559e5976c/RF24Network.cpp#L94).

SilentM4X, 2022-05-20:
I think you are right - i set up the system 3 years ago and due to the complexity of the setup I never dared to change anything. But I take this as a lesson. Thanks for the tip. This will be the game changer :)

And yes, i call `RF24::setRetries()` after begin:

```
radio = RF24(22, 0)
network = RF24Network(radio)
radio.begin()
time.sleep(0.1)
network.begin(90, this_node)
radio.setDataRate(RF24_250KBPS)
radio.setRetries(10,15)
radio.setCRCLength(RF24_CRC_8)
radio.enableDynamicPayloads()
```

2bndy5, 2022-05-20:
We have an updated setup script that should make your life easier. See [updated automated install instructions](https://github.com/nRF24/RF24/blob/master/docs/linux_install.md#automated-install)

SilentM4X, 2022-05-20:
Nice, thank you very much. I will try it in the next days :)

2bndy5, 2022-06-06:
ping

Can we close this? Is there additional need for help?

SilentM4X, 2022-06-07:
Hey, yes please and thank you :)

2bndy5, 2022-06-07:
If you have trouble building the python wrappers, feel free to ask on this thread or start a new one.

closed 2022-06-07 by 2bndy5