nRF24/RF24Network #197: NRF24 network node communication issue

OTeknoloji, 2022-11-28:
Hello everyone,

I am trying to do a project with NRF24. But somehow I can't receive data from all nodes. I have my main controller at level 0, a hub at level 1 and two devices at level 2. I am receiving and sending data from device 1, but the device cannot receive data at once. I would be glad if you help.

The codes are as follows:

[Node_00.txt](https://github.com/nRF24/RF24Network/files/10107663/Node_00.txt)

[Node_01.txt](https://github.com/nRF24/RF24Network/files/10107691/Node_01.txt)

[Node_011.txt](https://github.com/nRF24/RF24Network/files/10107696/Node_011.txt)

[Node_021.txt](https://github.com/nRF24/RF24Network/files/10107706/Node_021.txt)

2bndy5, 2022-11-28:
> but the device cannot receive data at once

Which device cannot receive?

2bndy5, 2022-11-28:
Just to be clear, the network level is related to the number of octal digits in the address, not the value of the octal digits in the address. So, for the nodes you describe, you would have a topology like so:
[image]

OTeknoloji, 2022-11-28:
Hi,

 device 021.  I do as this topology. 
https://nrf24.github.io/RF24Network/md_docs_tuning.html

I have two problems. One of them is that I get one node values at same time.
 Second  whatt does it mean this code 
#define EXTERNAL_DATA_MAX_SIZE MAX_PAYLOAD_SIZE

2bndy5, 2022-11-28:
I fixed my graph in my previous comment. I just noticed you didn't upload a code sample for node 02.

> I get one node values at same time

I don't know what you mean.

> Second whatt does it mean this code
> #define EXTERNAL_DATA_MAX_SIZE MAX_PAYLOAD_SIZE

That's not really library code. I'm guessing you are referring to the RX priority example: https://github.com/nRF24/RF24Network/blob/b8324b3cbd7bcbec307b3a13b7c7d9cf0f13e981/examples/Network_Priority_RX/Network_Priority_RX.ino#L67-L73

OTeknoloji, 2022-11-28:
yeah I am refering to RX priority.
that is, when I run two devices at the same time, I can't get data from both. but if I run 011 or 021 one by one, I can get the data from one device. I don't think there is a problem with aders. I can't use it at the same time for another reason I don't know.

OTeknoloji, 2022-11-28:
> BTW, the nRF24L01 cannot simultaneously receive and send data. Furthermore, it cannot receive multiple payloads simultaneously either; it has to receive payloads individually (one after another).

this is exactly what i want to do how can i do this

2bndy5, 2022-11-28:
In your code for node 021 you use [`RF24Network::read()`](https://nrf24.github.io/RF24Network/classRF24Network.html#a6630ab36b8a42cc376c925721f39188d), but you tell it to fetch a maximum of 0 bytes.
```cpp
  if (network.available()) {
    RF24NetworkHeader header;
    network.read(header, &dataBuffer, 0); // dataBuffer remains unchanged
  }
```

---
The lib is coded to retry for a number of milliseconds if sending fails. That is how sending data from many nodes to 1 node is handled. I think your code just needs to relax a little between transmissions. Without a deliberate delay between sending, a single node can be continuously spam the network and end up occupying the network while others are trying to send data.

OTeknoloji, 2022-11-28:
ok how can do this so min relax  to transmissions

2bndy5, 2022-11-28:
This is done in the RX Priority example using a set interval of milliseconds:
https://github.com/nRF24/RF24Network/blob/b8324b3cbd7bcbec307b3a13b7c7d9cf0f13e981/examples/Network_Priority_RX/Network_Priority_RX.ino#L109-L110

OTeknoloji, 2022-11-28:
ok thank you so much for your helping  I will try

OTeknoloji, 2022-11-29:
Hi I tried but it is not working  do you have any different idea?

closed 2022-11-29 by OTeknoloji