nRF24/RF24Network #198: how can clear or reset buffer?

OTeknoloji, 2022-11-29:
Hi everyone,

I set up a network of 4 devices. However, only the first 5 or 10 seconds are transferred, and then there is no data transfer. Maybe the buffer is over inflated. how can I reset it. Thank you your helping

network :
                                            l-----------> Node 011
Node00 l--------> Node01 
                                            l------------> Node021

Codes:
****************************************************************************
Node 00 : 
#include <RF24.h>
#include <RF24_config.h>
#include <nRF24L01.h>
#include <printf.h>
#include <RF24Network.h>
#include <RF24Network_config.h>

RF24 radio(7, 8);
RF24Network network(radio);

const uint16_t gaia = 00;
const uint16_t hermes = 01;
const uint16_t poseidon = 011;
const uint16_t demeter = 021;

int depo[4] = { 0, 1, 0, 1 };

float dem[4] = { 0, 0, 0, 0 };
float pose[4] = { 0, 0, 0, 0 };

uint32_t sendTimer = 0;
uint32_t userDataTimer = 0;

void setup() {
  Serial.begin(115200);
  radio.begin();
  printf_begin();  // needed for RF24* libs' internal printf() calls
  while (!Serial) {
    // some boards need this because of native USB capability
  }
  Serial.println(F("RF24Network/examples/Network_Separation_RX/"));

  if (!radio.begin()) {
    Serial.println(F("Radio hardware not responding!"));
    while (1) {
      // hold in infinite loop
    }
  }
  radio.setDataRate(RF24_250KBPS);
   radio.setPALevel(RF24_PA_LOW);
  network.begin(90, gaia);
  radio.printDetails();
}

void loop() {

  network.update();

  if (millis() - userDataTimer > 5000) {
    userDataTimer = millis();
 float veri[4] = { 0, 0, 0, 0 };
    while (network.available()) {
      RF24NetworkHeader header;
      network.read(header, &veri, sizeof(veri));
      if (header.from_node == demeter) {
        Serial.println("Demeter: ");
        for (int i = 0; i < 4; i++) {
          dem[i] = veri[i];
          Serial.println(dem[i]);
          Serial.print(" ");
        }
        Serial.println();
      } else if (header.from_node == poseidon) {
        Serial.println("Poseidon: ");
        for (int i = 0; i < 4; i++) {
          pose[i] = veri[i];
          Serial.println(pose[i]);
          Serial.print(" ");
        }
        Serial.println();
      }
    }
  }

 if (millis() - sendTimer > 1000) {
    sendTimer = millis();

    Serial.println(F("Sending data..."));
    RF24NetworkHeader header2(poseidon);
    bool ok = network.write(header2, &depo, sizeof(depo));
    Serial.println(ok ? F("OK") : F("Fail"));
 }  
}
*******************************************************************************
Node 01 : 
#include <RF24.h>
#include <RF24_config.h>
#include <nRF24L01.h>
#include <printf.h>
#include <RF24Network.h>
#include <RF24Network_config.h>

RF24 radio(7, 8);
RF24Network network(radio);

const uint16_t hermes = 01;

void setup() {
  Serial.begin(115200);
  radio.begin();
  printf_begin();  // needed for RF24* libs' internal printf() calls
  while (!Serial) {
    // some boards need this because of native USB capability
  }
  Serial.println(F("RF24Network/examples/Network_Separation_RX/"));

  if (!radio.begin()) {
    Serial.println(F("Radio hardware not responding!"));
    while (1) {
      // hold in infinite loop
    }
  }
  radio.setDataRate(RF24_250KBPS);
   radio.setPALevel(RF24_PA_LOW);
  network.begin(90, hermes);
  radio.printDetails();
}

void loop() {

  network.update();

 
}

**************************************
Node 011 : 
#include <RF24.h>
#include <RF24_config.h>
#include <nRF24L01.h>
#include <printf.h>
#include <RF24Network.h>
#include <RF24Network_config.h>

RF24 radio(7, 8);
RF24Network network(radio);

const uint16_t gaia = 00;
const uint16_t poseidon = 011;

int depo[4] = { 0, 0, 0, 0 };
float pose[4] = { 160, 80, 160, 80 };

uint32_t sendTimer = 0;
uint32_t userDataTimer = 0;

void setup() {
  Serial.begin(115200);
  radio.begin();
  printf_begin();  // needed for RF24* libs' internal printf() calls
  while (!Serial) {
    // some boards need this because of native USB capability
  }
  Serial.println(F("RF24Network/examples/Network_Separation_RX/"));

  if (!radio.begin()) {
    Serial.println(F("Radio hardware not responding!"));
    while (1) {
      // hold in infinite loop
    }
  }
  radio.setDataRate(RF24_250KBPS);
   radio.setPALevel(RF24_PA_LOW);
  network.begin(90, poseidon);
  radio.printDetails();
}

void loop() {

  network.update();

  
 if (millis() - userDataTimer > 5000) {
    userDataTimer = millis();

int veri[4] = { 0, 0, 0, 0 };
  while (network.available()) {
    RF24NetworkHeader header;

    network.read(header, &veri, sizeof(veri));
    Serial.println("depo: ");
    for (int i = 0; i < 4; i++) {
      depo[i] = veri[i];
      Serial.print(depo[i]);
      Serial.print(" ");
    }
    Serial.println();
  }
 }
  if (millis() - sendTimer > 1000) {
    sendTimer = millis();

    Serial.println(F("Sending data..."));
    RF24NetworkHeader header2(gaia);
    bool ok = network.write(header2, &pose, sizeof(pose));
    Serial.println(ok ? F("OK") : F("Fail"));
  }
}
**********************************************************************
Node 021 : 
#include <RF24.h>
#include <RF24_config.h>
#include <nRF24L01.h>
#include <printf.h>
#include <RF24Network.h>
#include <RF24Network_config.h>

RF24 radio(7, 8);
RF24Network network(radio);

const uint16_t gaia = 00;
const uint16_t demeter = 021;

float dem[4] = { 1, 80, 23, 65 };

uint32_t sendTimer = 0;

void setup() {
  Serial.begin(115200);
  radio.begin();
  printf_begin();  // needed for RF24* libs' internal printf() calls
  while (!Serial) {
    // some boards need this because of native USB capability
  }
  Serial.println(F("RF24Network/examples/Network_Separation_RX/"));

  if (!radio.begin()) {
    Serial.println(F("Radio hardware not responding!"));
    while (1) {
      // hold in infinite loop
    }
  }
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_LOW);
  network.begin(90, demeter);
  radio.printDetails();
}

void loop() {

  network.update();

  if (millis() - sendTimer > 1000) {
    sendTimer = millis();

    Serial.println(F("Sending data..."));
    RF24NetworkHeader header(gaia);
    bool ok = network.write(header, &dem, sizeof(dem));
    Serial.println(ok ? F("OK") : F("Fail"));
    
  }
}

2bndy5, 2022-11-30:
What buffer? All the arrays that you use are iterated with a `for` loop. This seems like a generic C++ question because there are a few ways to reset an array's contents to 0 (or whatever desired init value is).

It is best to not assume what the problem is before seeking help. When you open an issue, you should include as much information about the problem you are experiencing.

2bndy5, 2022-11-30:
It is best to make sure `network.available()` returns false before calling `network.update()` again. Otherwise any incoming payloads will not get queued if the lib's internal queue is full.
```cpp
  network.update();

/* don't wait to handle the incoming payloads! */
// if (millis() - userDataTimer > 5000) {
//    userDataTimer = millis();

  int veri[4] = { 0, 0, 0, 0 };
  while (network.available()) {
```
Resetting the lib's queue is not a good idea because it means data loss.

OTeknoloji, 2022-11-30:
when i start the device i get a certain number of data. but then i am not getting data.. I dont know Why.

and printing here:

RF24Network/examples/Network_Separation_RX/
SPI Speedz	= 10 Mhz
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	= 0x33 0xce 0x3e 0xe3
TX_ADDR		= 0xcccccc3ce3
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3e
EN_RXADDR	= 0x3f
RF_CH		= 0x5a
RF_SETUP	= 0x23
CONFIG		= 0x0f
DYNPD/FEATURE	= 0x3f 0x04
Data Rate	= 250 KBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 0
Demeter: 
1 80 23 65 
Sending data...
OK
Poseidon: 
250 80 160 80 
Sending data...
OK
Demeter: 
1 80 23 65 
Sending data...
OK
Poseidon: 
250 80 160 80 
Sending data...
OK
Demeter: 
1 80 23 65 
Sending data...
OK
Poseidon: 
250 80 160 80 
Sending data...
OK
Demeter: 
1 80 23 65 
Sending data...
OK
Poseidon: 
250 80 160 80 
Sending data...
OK
Sending data...
OK
Sending data...
OK
Sending data...
OK
Sending data...
OK
Sending data...
OK
Sending data...
..........
...........

OTeknoloji, 2022-11-30:
Bro 
you are greatfull 

I changed code this and it is working, but ı want to ask a question. if I used "float"  veri[ 4] it is not working but if I used ınt16_t veri[4] ,it is working. So mean ı want to send float data for example  150.76

 while (network.available()) {
      RF24NetworkHeader header;
      int16_t veri[4] = { 0, 0, 0, 0 };
      network.read(header, &veri, sizeof(veri));

2bndy5, 2022-11-30:
node 00 is sending `int depo[4]` to node 011, so the expected data type is an array of 4 integers.

OTeknoloji, 2022-11-30:
this code for your says "Resetting the lib's queue is not a good idea because it means data loss."

while (network.available()) {
RF24NetworkHeader header;
int16_t veri[4] = { 0, 0, 0, 0 };
network.read(header, &veri, sizeof(veri));

I mean it is ok it is working.

closed 2022-11-30 by OTeknoloji