nRF24/RF24Network #199: ESP32 not receiving correct data from NRF24L01

rptmatrox, 2022-12-29:
First of all: Im sorry for my english. I´ll try to do my best. =)

Well, the story goes like this:
I have a homemade project for monitoring various parameters at home such as: temperature, humidity, atmospheric pressure, current, voltage, power, etc...
In the first version I used an Arduino MEGA 2560 (WEMOS WIFI) as a base and arduinos NANO as satellites with sensors. The data was sent to the BLYNK platform.
The routine is: every 15 seconds the base send a data request for each satellite and when they receive it, they send the sensor data back to the base. This way I would have a synchronized timestamp of all collected data.

**The first problem (SOLVED):** over time I had issues with the WIFI connection on the MEGA base and I opted to exchange it for an ESP32.
**The second problem (NOT SOLVED):** now I have a stable WIFI connection, but now I have other issue receiving data from the NRF24L01.

The code of NANOs is the same.
The code of BASE (MEGA vs ESP32) is almost the same except for the CE, CSN PIN configuration.

**To simplify the example and make the code cleaner I removed all the BLYNK integration, Wi-Fi, and some satellites and left only one set to you help me with the issue.**

### More details below:

### BASE CODE (ARDUINO MEGA and ESP32) *The Diference is the CE/ CSN Pin declaration comment or not:
(CODE IN TXT FORMAT BELOW)
[RF_NETWORK_MEGA_BASE_ESP32_TESTE.txt](https://github.com/nRF24/RF24Network/files/10319955/RF_NETWORK_MEGA_BASE_ESP32_TESTE.txt)

### ARDUINO NANO CODE (SAT2 reading temperature and Humidity):
(CODE IN TXT FORMAT BELOW)
[RF_NETWORK_NANO_SAT2.txt](https://github.com/nRF24/RF24Network/files/10319948/RF_NETWORK_NANO_SAT2.txt)

OK, the Serial results for each board:
ARDUINO NANO - The sensors are reading correct values and the PAYLOAD is OK. (Since using MEGA as BASE the value is CORRECT):
[image]

ARDUINO MEGA - The Value received is OK:
[image]

ESP32 - The Value received is NOT OK:
[image]

As you can see there is no print of the NRF24L01 parameters and the payload received appear to be "trash".

### The hardware PIN configuration in ESP32 are:

GND..........GND
VCC 3.3....VCC 3.3
MOSI........D23 (2)
MISO........D19 (8)
SCK...........D18 (9)
CSN..........D5  (10)
CE.............D4  (13)
IRQ...........unplugged

### Thank you very much for your attention and I hope to have enough information to help me solve this problem.

Avamander, 2022-12-29:
Are the integers in the struct the same size on both platforms and aligned identical in memory? (Using fixed-size data types and telling the compiler to *pack* the struct would be a really really good thing to so)

Are you calculating the size of the payloads correctly on both ends?

You can determine if the bug is in your code by taking the example sketches and by running those without modifications.

rptmatrox, 2022-12-29:
@Avamander Thanks for your response.
After your comment I researched about it and found this topic that spoke exactly about this variable type issue:
https://forum.arduino.cc/t/data-transmision-issue-from-arduino-pro-mini-to-esp32-via-nrf24l01/918278/4

I tested in my code change the variables inside STRUCT to int32_t in both codes (TX and RX) and it worked. 
Like this:

```
struct payload_R_SAT2 {
  int32_t temperaturaI;
  int32_t umidadeI;
};
```

Thanks a LOT for the TIP.
Anyway this topic could be usefull for other users =)

Avamander, 2022-12-29:
Great to hear it worked! I really recommend reading about and using `__attribute__((packed))` as well.

closed 2022-12-29 by Avamander

rptmatrox, 2022-12-29:
> Great to hear it worked! I really recommend reading about and using `__attribute__((packed))` as well.

Thank you again. I'll find out.