nRF24/RF24Network #201: how to support 8 pipes (on nRF5x devices)

2bndy5, 2023-01-03:
Concerning future ventures with the nRF5x ESB, I was thinking that 8 pipes shouldn't be too hard to support.

- [ ] pipe/address validation will have to conditionally forego `x <= 5` presumption about 6 pipes
- [ ] mesh master needs a way to know when a child node can have 8 pipes. This will likely be the most costly in code size. We could augment the mesh handshake info to additionally contain data that indicates to master the number of pipes supported. Then we have to figure out how this data will fit into `DHCP()`.
- [ ] figure out 2 new preamble-friendly physical address bytes default value (for pipes 6 & 7).

If I'm missing something let me know. I would be curious to play with this lib using 8 pipes, so I'm hoping its possible.

2bndy5, 2023-01-03:
There is still some reserved space in the network sys codes, so we could add a `NETWORK_NODE_PROPERTIES` sys code to the network layer protocol. This sys code could then be used to confirm the connection after master has leased the address. My main concern is when this feature is used with a node running an older version of the lib.

TMRh20, 2023-01-03:
I've given this some thought and with the increase in code size etc, I'm not sure its a good idea. RF24Mesh will still compile and run on ATTiny, even though you can't do much with it, and I would like to keep it that way. 

I think on the third point, it might be kind of difficult to support a mixed network if some use differing amounts of pipes. It would be way easier to either have an nrf24l01 or nrf52840 network.

TMRh20, 2023-01-03:
That would work. It would be interesting to get it working with a mixed network but again not sure what it would take.

So, what if all code that supports 8 pipes is #ifdef out? That way the same lib cab be used for different networks.

TMRh20, 2024-06-14:
I think we have the first two items picked off, its just a matter of if we want to support mixed networks. I don't really want to lol.

TMRh20, 2026-10-23:
I think this issue can be closed if we are both in agreement?

2bndy5, 2026-10-23:
Oops. Yeah, should closed this years ago.

closed 2026-10-23 by 2bndy5