nRF24/RF24Network #205: Device can send or receive only depending on wiring

davidmritter, 2023-04-12:
I have four Pi Picos attached to four nRF24L01+ radios. Depending on if I have them wired directly to the Picos with jumpers or if I use a breakout board, the module will only be able to send or receive data respectively. I have no idea why this is the case, I have swapped to the code between the four setups and unless the sender is attached to the Pico with jumpers, and the receiver is using a breakout board the code fails to run.

Any idea as to why this is happening? I am hoping to get these on to protoboards so I will need to use breakout boards but I can't move the transmitter due to this.

2bndy5, 2023-04-12:
The adapter boards typically have voltage regulators, so it is likely that the radios aren't getting enough power from the 3v supply line. Maybe I read that wrong...

davidmritter, 2023-04-12:
In this case, I have two different adapter boards. One with a regulator, and one without.

If I use either breakout board, it can only receive data, the other board will never successfully transmit data. If I use a board that is directly wired to the pico(header pins -> jumper wire -> RF24 board) it can send data and will rarely receive it( only in tests with both directly attached). I have swapped out all of the parts and double checked all the wiring but this is the result I consistently get.

I also have tested to see if the power on order has any effect, but it does not. The only variable is how the RF boards are wired up.

2bndy5, 2023-04-12:
Strictly speaking, if a RF24 module fails to transmit, then that usually means there is something up with the power lines to the radio. Transmitting (including ACK packets) requires the most current, so this is a big hint to me.

Are there capacitors at play here?

davidmritter, 2023-04-12:
When I first tested, only the breakout board without a regulator had a 10uF cap added, none of the others did. Adding a 10uF cap has worked on one of the directly connected boards(it now receives most of the packets), but has had no impact on the other one(it can still only send).

2bndy5, 2023-04-13:
To be clear, if it has a voltage regulator, are you feeding it 5v? Usually the regulator is meant to provide a stable 3v from a 5v input.

> has had no impact on the other one(it can still only send).

If it sends successfully with auto-ack enabled, then it can receive. Maybe check to make sure the time interval used for receiving aligns with the other nodes' time interval used for transmitting.

Are you using a RF24 lib example to test?

davidmritter, 2023-04-13:
I am using 5 volts with the regulator.
I was using my own code that doesn't check the ack packet. Using the helloworld_rx/tx example, I either get nothing or the receiver sees a packet, but the sender prints out a failed message. I do have to modify the example to turn down the data rate to 250KBPS or it fails to work.

Here is my own code I have been testing with:
```
#include <RF24.h>
#include <RF24Network.h>

// Define the radio pins
RF24 radio(20, 17); // CE, CSN
RF24Network network(radio);

const uint16_t living_room = 01;
unsigned long last_request_time = 0;
unsigned long request_interval = 10000; // 10 seconds

void setup() {
    Serial.begin(115200);
    while (!Serial) {
      // some boards need this because of native USB capability
    }
    Serial.println("Starting up");
    radio.begin();
    radio.setPALevel(RF24_PA_MAX);
    radio.setDataRate(RF24_250KBPS);
    network.begin(/*channel*/ 80, /*node address*/ 00);
    radio.printDetails();
}

void loop() {
    network.update();
    unsigned long current_time = millis();

    if (current_time - last_request_time > request_interval) {
        last_request_time = current_time;
        requestTemperature(living_room);
        Serial.println("Asking for temp");
    }

    while (network.available()) {
        RF24NetworkHeader header;
        float temperature;
        network.read(header, &temperature, sizeof(temperature));

        Serial.print("Temperature from Node ");
        Serial.print(header.from_node, OCT);
        Serial.print(": ");
        Serial.print(temperature);
        Serial.println(" °C");
    }
}

void requestTemperature(uint16_t node_address) {
    RF24NetworkHeader header(node_address, 'T');
    network.write(header, 0, 0);
}
```
and for the other unit:
```
#include <RF24.h>
#include <RF24Network.h>

// Define the radio pins
RF24 radio(20, 17); // CE, CSN
RF24Network network(radio);

const uint16_t living_room = 01;
unsigned long last_request_time = 0;
unsigned long request_interval = 10000; // 10 seconds

void setup() {
    Serial.begin(115200);
    while (!Serial) {
      // some boards need this because of native USB capability
    }
    Serial.println("Starting up");
    radio.begin();
    radio.setPALevel(RF24_PA_MAX);
    radio.setDataRate(RF24_250KBPS);
    network.begin(/*channel*/ 80, /*node address*/ 00);
    radio.printDetails();
}

void loop() {
    network.update();
    unsigned long current_time = millis();

    if (current_time - last_request_time > request_interval) {
        last_request_time = current_time;
        requestTemperature(living_room);
        Serial.println("Asking for temp");
    }

    while (network.available()) {
        RF24NetworkHeader header;
        float temperature;
        network.read(header, &temperature, sizeof(temperature));

        Serial.print("Temperature from Node ");
        Serial.print(header.from_node, OCT);
        Serial.print(": ");
        Serial.print(temperature);
        Serial.println(" °C");
    }
}

void requestTemperature(uint16_t node_address) {
    RF24NetworkHeader header(node_address, 'T');
    network.write(header, 0, 0);
}
```

I had a temp sensor connected to one of the boards, I removed it to see if it had any effect, but it didn't.

2bndy5, 2023-04-13:
Sorry, I forgot this was a RF24Network issue. Have you tried any RF24 examples to ensure basic radio functionality works?

> the receiver sees a packet, but the sender prints out a failed message

This is indicative of another node failing to transmit the auto-ack response. Another reason why we should check that all radios satisfy RF24 lib functionality. What radio modules are you using?

davidmritter, 2023-04-13:
I swapped to the "Getting Started" example, edited to match the CE and CSN pins and got the following:
Node 0:
```
20:07:39.347 -> Which radio is this? Enter '0' or '1'. Defaults to '0'
20:08:05.364 -> radioNumber = 0
20:08:05.364 -> *** PRESS 'T' to begin transmitting to the other node
20:08:33.462 -> Received 4 bytes on pipe 1: 0.00
20:08:36.533 -> Received 4 bytes on pipe 1: 0.01
20:08:37.569 -> Received 4 bytes on pipe 1: 0.02
20:08:38.569 -> Received 4 bytes on pipe 1: 0.03
20:08:39.600 -> Received 4 bytes on pipe 1: 0.04
20:08:40.604 -> Received 4 bytes on pipe 1: 0.04
20:08:42.641 -> Received 4 bytes on pipe 1: 0.05
20:08:44.676 -> Received 4 bytes on pipe 1: 0.06
20:08:46.744 -> Received 4 bytes on pipe 1: 0.06
20:08:47.747 -> Received 4 bytes on pipe 1: 0.06
20:08:49.814 -> Received 4 bytes on pipe 1: 0.06
20:08:51.852 -> Received 4 bytes on pipe 1: 0.07
20:08:55.924 -> Received 4 bytes on pipe 1: 0.08
20:08:56.957 -> Received 4 bytes on pipe 1: 0.09
20:08:57.962 -> Received 4 bytes on pipe 1: 0.10
20:09:00.029 -> Received 4 bytes on pipe 1: 0.10
20:09:03.097 -> Received 4 bytes on pipe 1: 0.11
20:09:04.100 -> Received 4 bytes on pipe 1: 0.12
20:09:06.169 -> Received 4 bytes on pipe 1: 0.12
20:09:08.229 -> Received 4 bytes on pipe 1: 0.12
20:09:09.230 -> Received 4 bytes on pipe 1: 0.12
20:09:12.299 -> Received 4 bytes on pipe 1: 0.13
20:09:18.658 -> *** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
20:09:18.658 -> Transmission failed or timed out
20:09:19.693 -> Transmission failed or timed out
20:09:20.691 -> Transmission successful! Time to transmit = 5756 us. Sent: 0.13
20:09:21.724 -> Transmission failed or timed out
20:09:22.759 -> Transmission failed or timed out
20:09:23.792 -> Transmission successful! Time to transmit = 24849 us. Sent: 0.14
20:09:24.825 -> Transmission successful! Time to transmit = 23111 us. Sent: 0.15
20:09:25.827 -> Transmission failed or timed out
20:09:26.863 -> Transmission successful! Time to transmit = 10946 us. Sent: 0.16
20:09:27.864 -> Transmission successful! Time to transmit = 2258 us. Sent: 0.17
20:09:28.866 -> Transmission failed or timed out
20:09:29.902 -> Transmission failed or timed out
20:09:30.934 -> Transmission failed or timed out
20:09:31.968 -> Transmission failed or timed out
20:09:33.004 -> Transmission failed or timed out
20:09:34.034 -> Transmission successful! Time to transmit = 26588 us. Sent: 0.18
20:09:35.036 -> Transmission successful! Time to transmit = 2257 us. Sent: 0.19
20:09:36.035 -> Transmission failed or timed out
20:09:37.069 -> Transmission failed or timed out
20:09:38.070 -> Transmission successful! Time to transmit = 3992 us. Sent: 0.20
20:09:39.106 -> Transmission failed or timed out
20:09:40.140 -> Transmission failed or timed out
20:09:41.173 -> Transmission failed or timed out
20:09:42.210 -> Transmission failed or timed out
20:09:43.211 -> Transmission failed or timed out
20:09:44.246 -> Transmission successful! Time to transmit = 7481 us. Sent: 0.21
20:09:45.246 -> Transmission failed or timed out
20:09:46.278 -> Transmission failed or timed out
20:09:47.311 -> Transmission failed or timed out
20:09:48.343 -> Transmission failed or timed out
20:09:49.343 -> Transmission successful! Time to transmit = 521 us. Sent: 0.22
20:09:50.345 -> Transmission successful! Time to transmit = 2270 us. Sent: 0.23
20:09:51.381 -> Transmission failed or timed out
20:09:52.383 -> Transmission successful! Time to transmit = 5732 us. Sent: 0.24
20:09:53.385 -> Transmission successful! Time to transmit = 525 us. Sent: 0.25
20:09:54.418 -> Transmission failed or timed out
20:09:55.422 -> Transmission successful! Time to transmit = 23106 us. Sent: 0.26
20:09:56.423 -> Transmission successful! Time to transmit = 5753 us. Sent: 0.27
20:09:57.458 -> Transmission failed or timed out
20:09:58.462 -> Transmission successful! Time to transmit = 524 us. Sent: 0.28
20:09:59.496 -> Transmission successful! Time to transmit = 23115 us. Sent: 0.29
[... 88 lines cut ...]
20:09:49.343 -> Received 4 bytes on pipe 1: 0.22
20:09:50.345 -> Received 4 bytes on pipe 1: 0.23
20:09:51.347 -> Received 4 bytes on pipe 1: 0.24
20:09:52.380 -> Received 4 bytes on pipe 1: 0.24
20:09:53.385 -> Received 4 bytes on pipe 1: 0.25
20:09:54.386 -> Received 4 bytes on pipe 1: 0.26
20:09:55.422 -> Received 4 bytes on pipe 1: 0.26
20:09:56.423 -> Received 4 bytes on pipe 1: 0.27
20:09:57.426 -> Received 4 bytes on pipe 1: 0.28
20:09:58.462 -> Received 4 bytes on pipe 1: 0.28
20:09:59.463 -> Received 4 bytes on pipe 1: 0.29
20:10:00.496 -> Received 4 bytes on pipe 1: 0.30
20:10:01.498 -> Received 4 bytes on pipe 1: 0.31
20:10:02.533 -> Received 4 bytes on pipe 1: 0.32
20:10:03.533 -> Received 4 bytes on pipe 1: 0.33
20:10:04.536 -> Received 4 bytes on pipe 1: 0.34
20:10:05.571 -> Received 4 bytes on pipe 1: 0.34
20:10:06.575 -> Received 4 bytes on pipe 1: 0.35
20:10:07.574 -> Received 4 bytes on pipe 1: 0.36
20:10:08.604 -> Received 4 bytes on pipe 1: 0.36
20:10:09.636 -> Received 4 bytes on pipe 1: 0.36
20:10:10.668 -> Received 4 bytes on pipe 1: 0.36
```

One thing I noticed is that node one was far more reliable in receiving the packets then node zero for some reason. As for the model they are nRF24L01+ modules.

davidmritter, 2023-04-13:
I swapped out the setup, the direct attached I plugged into a Pico W I have lying around, and the paired it up with the powered breakout on a standard breakout board. That seems to be working perfectly with the Getting Started code so I think you were right on the power draw being the issue. I do find it a bit odd though, I would think that it should have enough power given that these modules shouldn't be demanding much current.

2bndy5, 2023-04-13:
Check for ambient noise (RF24 scanner example)? Honestly, if this isn't a power issue, then I don't know what else to check.

Last I checked (before the shortage), the pico boards use a switching 3v regulator, so there's likely a lot of noise (due to the nature of switching regulators) in the 3v power lines that needs to be filtered out with capacitors.

2bndy5, 2023-04-13:
Maybe try introducing a small capacitor and a larger capacitor in parallel to the radio's VCC and GND pins. Fitting noise is best done with multiple varying capacitance in parallel...

davidmritter, 2023-04-13:
I will give that a try. I have also run the scanner, it looks like I am mostly in the clear. Just on section used up by the neighbors WiFi and a bit of noise occasionally on other channels. Even in the range being used by that network, I never see higher then a 4 and most of the time they are 1 or 2 so I may be able to use them for some short range stuff.

I'll play around with the capacitors and see how it goes. If/when I find some that work I'll post back here.

Avamander, 2023-04-13:
Noise could of course be both high- and low-frequency, so it might take both low ESR tantalum/ceramic and (higher ESR) electrolytic capacitors to fully filter out.

davidmritter, 2023-04-15:
I played around a bit and found that using a 10uF and 100nF capacitors in parallel worked the best, though I still only got a 75% receive success and about a 90% transmit success to a unit using an external regulator. I remember reading something about the Pico's power supply having two different operating modes so I will look further into that. For now though it seems that you can't reliably power a NRF24L01+ from the Pico directly. Thanks @2bndy5 and @Avamander for the help.

closed 2023-04-15 by davidmritter