nRF24/RF24Network #206: Connect to more than 5 devices

luke2023, 2023-05-04:
Hello ,thank you for your library!I found this tutorial very helpful about your library https://howtomechatronics.com/tutorials/arduino/how-to-build-an-arduino-wireless-network-with-multiple-nrf24l01-modules/

In the intro is says it is limited to 5 devices each module, but I don’t see devices limitation on your examples.
If the limitation is correct ,how do I modify the code to make a single device connect to more than 5 devices?
Because I have many devices in a small area and they might be unstable if I use so many levels (if the battery is out or something).

Thank you so much!
Luke lin

TMRh20, 2023-05-04:
The radios are designed with a limited number of 'pipes' that can be assigned addresses. Typically, if you want to have more than 5 devices connected to a single device, you need to create a network with intermediate nodes passing traffic to more distant nodes, or you can disable Auto-Ack and have all of the devices using the same address and use multicast. These are the two most common scenarios.

2bndy5, 2023-05-04:
It is a hardware limitation, not a software kimitation.

2bndy5, 2023-05-04:
Please don't use the `RF24 radio` object directly when using RF24Network library! It will break the `network` object's behavior if you manually set the addresses yourself, especially if you don't know what exactly you're doing.

TMRh20 is referring to [`RF24Network::multicast()`](https://nrf24.github.io/RF24Network/classRF24Network.html#a96a66907061c54ee1fea6ea7f78d6827) which is listed in the docs as an advanced function because it requires a basic understanding of network topology (see [here](https://nrf24.github.io/RF24Network/md_docs_addressing.html) and [here](https://pyrf24.readthedocs.io/en/latest/topology.html))

2bndy5, 2023-05-04:
It should be noted that _network addresses_ (AKA "logical addresses") are different from the physical addresses used for the radios' pipes.

luke2023, 2023-05-04:
@2bndy5 thank you so much for your answer ! I  will share my project on github when I am done !  (making a low cost IOT standard for

2bndy5, 2023-05-04:
> or you can disable Auto-Ack and have all of the devices using the same address and use multicast.

example pseudo code:
```cpp
const uint16_t siblings = 01; // used for all devices except master
const uint16_t master_node = 00;

network.begin(siblings); // pass `master_node` on the network master device 
radio.setAutoAck(false); // not a usually recommended

// ...

// send to all siblings from master node
network.multicast(header, message, sizeof(message), /*level*/ 1);
// 1 is the network level for devices using the logical address `01`-`05`
```

TMRh20, 2023-05-06:
I guess I was thinking about using the RF24 layer when I made the comment about disabling auto-ack and using the same address. Typically with RF24Network, you can also just call `network.multicastLevel(1)` on each device while retaining normal functionality and having a unique id for each device. Both methods will work I guess.

2bndy5, 2023-05-06:
Yeah, that's a much better idea that should retain normal network behavior. Here's an updated example pseudo code:

```cpp
network.begin(thisNode); // pass logical address specific to the device
network.setMulticastLevel(1); // for all devices except master

// ...

// send to all siblings from master node
network.multicast(header, message, sizeof(message), /*level*/ 1);
```

2bndy5, 2023-05-06:
Disabling AutoAck with the `radio` object isn't guaranteed to persist after a normal `network.write()`. Furthermore, calling `radio.setAutoAck(false)` before `network.write()` could instigate data loss because network-acks is disabled as well.

luke2023, 2023-05-06:
@2bndy5 thank you for the replay!
If I can remain each unique id for each device(like TMRh20 recommended) ,can 012 still get the message while 01 has dead?
Thank you!

Luke Lin

TMRh20, 2023-05-06:
> If I can remain each unique id for each device(like TMRh20 recommended) ,can 011 get the message with network.multicast( ,, /*level*/ 1); while 01 has dead?

Yes, as long as you call `network.multicastLevel(1);` on each device except master. Master can be level 0.

> Or I should set every device to 01 except master?

You should set the devices as normal, each with a unique address. Then call `network.multicast(header, message, sizeof(message), /*level*/ 1);` when sending

luke2023, 2023-05-07:
@TMRh20  Thank you, if I want to broadcast to all the devices, should I use 
`For (int i =0 ;i<5;i++){network.multicast( ,, /*level*/ i);} `? Thank you!

2bndy5, 2023-05-07:
Yes, if you want to use the default multicast levels that are calculated when you pass the logical address to `network.begin()`.

If you set each device to use the same multicast level with [`network.multicastLevel(x)`](https://nrf24.github.io/RF24Network/classRF24Network.html#a4fb7c61eee75d708a4ba20bb1caec54f), then you only need to call `network.multicast(..., /*level*/ x)` once.

closed 2023-05-08 by luke2023