nRF24/RF24Network #208: How to support NRF52 library?

TMRh20, 2023-05-25:
With the running NRF24 + BLE or NRF52 library both needing to be supported on nrf52840, I'm not sure how to best incorporate the changes needed for NRF52 lib nrf_to_nrf. This applies to RF24Network and RF24Mesh.

I think it is a more likely scenario for users to use the onboard radio for everything they are doing, but some may want to run a nRF24 module along with BLE or something. 

How to incorporate both models into RF24?

https://github.com/nRF24/RF24Network/tree/nrf52Support

https://github.com/nRF24/RF24Mesh/tree/NRF25Support

2bndy5, 2023-05-25:
Thanks for setting up those branches.

I started #204 to propose a way to do this. My main concern is maintaining backward compatibility. It wouldn't be a problem if nrf_to_nrf had concurrent protocol support, but that's probably more problematic than using dedicated hardware for each protocol.

2bndy5, 2023-05-30:
> This applies to RF24Network and RF24Mesh.

Should this be considered a duplicate of #204? Is there a reason to discuss these layers separately?

closed 2023-05-30 by TMRh20