nRF24/RF24Network #212: Unable to use the RF24 libraries with platformio since the v1.0.18 release

shyney7, 2023-06-20:
Since the new release of the RF24 libraries Im unable to use them with platformio anymore.
Even if I try to use the older version that was working for projects before this update.
Here is the error that I get: https://github.com/platformio/platformio-home/issues/5848

I know that until this point its a platformio issue thats why I opened it there instead of here, but even if I try to download every library manually and insert them into the lib folder Im getting a bunch of compiler errors.
This issue seems only to affect the RF24Network and RF24Mesh libraries. The RF24 library can be isntalled in platformio without problems.

This seems to only effect the nrf_to_nrf.cpp source file:

<details><summary>Details</summary>
<p>

```Processing lilygo-t-display-s3 (platform: espressif32; board: lilygo-t-display-s3; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lilygo-t-display-s3.html
PLATFORM: Espressif 32 (6.2.0) > LilyGo T-Display-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20008.0 (2.0.8)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RF24 @ 1.4.7        
|-- RF24Mesh @ 1.1.9    
|-- RF24Network @ 1.0.18
|-- SPI @ 2.0.0
Building in release mode
Compiling .pio\build\lilygo-t-display-s3\liba8a\nrf_to_nrf\nrf_to_nrf.cpp.o
Compiling .pio\build\lilygo-t-display-s3\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\lilygo-t-display-s3\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\lilygo-t-display-s3\FrameworkArduino\main.cpp.o
Compiling .pio\build\lilygo-t-display-s3\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\lilygo-t-display-s3\FrameworkArduino\wiring_pulse.c.o
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp: In function 'uint32_t addr_conv(const uint8_t*)':
Compiling .pio\build\lilygo-t-display-s3\FrameworkArduino\wiring_shift.c.o
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:22:12: error: '__REV' was not declared in this scope
     return __REV(
            ^~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp: In member function 'bool nrf_to_nrf::begin()':
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:86:5: error: 'NRF_CLOCK' was not declared in this scope
     NRF_CLOCK->EVENTS_HFCLKSTARTED = 0;
     ^~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:86:5: note: suggested alternative: 'F_LOCK'
     NRF_CLOCK->EVENTS_HFCLKSTARTED = 0;
     ^~~~~~~~~
     F_LOCK   
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:94:28: error: 'CLOCK_LFCLKSRC_SRC_Xtal' was not declared in this scope
     NRF_CLOCK->LFCLKSRC = (CLOCK_LFCLKSRC_SRC_Xtal << CLOCK_LFCLKSRC_SRC_Pos);
                            ^~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:94:55: error: 'CLOCK_LFCLKSRC_SRC_Pos' was not declared in this scope
     NRF_CLOCK->LFCLKSRC = (CLOCK_LFCLKSRC_SRC_Xtal << CLOCK_LFCLKSRC_SRC_Pos);
                                                       ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:102:5: error: 'NRF_RADIO' was not declared in this scope
     NRF_RADIO->POWER = 1;
     ^~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:102:5: note: suggested alternative: 'NRF_PA_LOW'
     NRF_RADIO->POWER = 1;
     ^~~~~~~~~
     NRF_PA_LOW
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:105:30: error: 'RADIO_PCNF0_S0LEN_Pos' was not declared in this scope
     NRF_RADIO->PCNF0 = (1 << RADIO_PCNF0_S0LEN_Pos) | (0 << RADIO_PCNF0_LFLEN_Pos) | (1 << RADIO_PCNF0_S1LEN_Pos);
                              ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:105:61: error: 'RADIO_PCNF0_LFLEN_Pos' was not declared in this scope
     NRF_RADIO->PCNF0 = (1 << RADIO_PCNF0_S0LEN_Pos) | (0 << RADIO_PCNF0_LFLEN_Pos) | (1 << RADIO_PCNF0_S1LEN_Pos);
[... 512 lines cut ...]
     NRF_RADIO->PCNF1 = (RADIO_PCNF1_WHITEEN_Disabled << RADIO_PCNF1_WHITEEN_Pos) | (RADIO_PCNF1_ENDIAN_Big << RADIO_PCNF1_ENDIAN_Pos) | (a_width - 1 << RADIO_PCNF1_BALEN_Pos) | (pSize << RADIO_PCNF1_STATLEN_Pos) | (staticPayloadSize << RADIO_PCNF1_MAXLEN_Pos);

                           ^~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:1035:237: error: 'RADIO_PCNF1_MAXLEN_Pos' was not declared in this scope
     NRF_RADIO->PCNF1 = (RADIO_PCNF1_WHITEEN_Disabled << RADIO_PCNF1_WHITEEN_Pos) | (RADIO_PCNF1_ENDIAN_Big << RADIO_PCNF1_ENDIAN_Pos) | (a_width - 1 << RADIO_PCNF1_BALEN_Pos) | (pSize << RADIO_PCNF1_STATLEN_Pos) | (staticPayloadSize << RADIO_PCNF1_MAXLEN_Pos);

                                                                            ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp: In member function 'void nrf_to_nrf::printDetails()':
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:1045:20: error: 'NRF_RADIO' was not declared in this scope
     Serial.println(NRF_RADIO->STATE);
                    ^~~~~~~~~
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:1045:20: note: suggested alternative: 'NRF_PA_LOW'
     Serial.println(NRF_RADIO->STATE);
                    ^~~~~~~~~
                    NRF_PA_LOW
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp: In member function 'nrf_crclength_e nrf_to_nrf::getCRCLength()':
.pio/libdeps/lilygo-t-display-s3/nrf_to_nrf/src/nrf_to_nrf.cpp:983:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
*** [.pio\build\lilygo-t-display-s3\liba8a\nrf_to_nrf\nrf_to_nrf.cpp.o] Error 1
================================================================== [FAILED] Took 9.20 seconds ==================================================================
```

</p>
</details>

TMRh20, 2023-06-20:
@2bndy5 OK I removed the dependancies from nrf_to_nrf, but am still getting errors installing RF24Mesh in platform io:

```
Libraries: Could not install library
PIO Core Call Error: "Resolving nanoatmega328 dependencies...\r\nLibrary Manager: RF24Mesh@1.1.9 is already installed\r\nLibrary Manager: Installing git+https://github.com/nRF24/RF24Network.git#nrf52Support\r\ngit version 2.37.0.windows.1\n\n\nCloning into 'C:\\Users\\Owner\\.platformio\\.cache\\tmp\\pkg-installing-4q2iap8o'...\nwarning: Could not find remote branch nrf52Support to clone.\nfatal: Remote branch nrf52Support not found in upstream origin\nError: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'nrf52Support', 'https://github.com/nRF24/RF24Network.git', 'C:\\\\Users\\\\Owner\\\\.platformio\\\\.cache\\\\tmp\\\\pkg-installing-4q2iap8o']"
```

Maybe it takes a bit to update fully? I dunno...

2bndy5, 2023-06-20:
It might take a bit for PIO lib registry to use the new release. 🤷🏼 I'll look into this and the open PRs tonight.

2bndy5, 2023-06-21:
I think we need to add the following patch to v1.x branch and retro release:
```diff
--- a/library.json
+++ b/library.json
@@ -17,7 +17,8 @@
         {
             "name": "nrf_to_nrf",
             "authors": "TMRh20",
-            "frameworks": "arduino"
+            "frameworks": "arduino",
+            "platforms":["nordicnrf52"]
         }
     ],
     "export":
```
Currently, RF24Network v1.0.18 will note nrf_to_nrf as a dependency for platforms other than `nordcnrf52`. While the removal of deps from nrf_to_nrf in v1.2.5 does avoid the erroneous dependency-loop, the compiler still tries to compile nrf_to_nrf for incompatible platforms because there is no `platforms` stipulation on the nrf_to_nrf dep.

My local attempts (compiling for esp32 board as done in  OP):
✅ RF24Network v1.0.17 & RF24Mesh v1.1.9
💥 RF24Network v1.0.18 & RF24Mesh v1.1.9
✅ RF24Network v2.0 (master) & RF24Mesh v2.0 (template-attempt2 branch)

2bndy5, 2023-06-21:
Just published RF24Network v1.0.19 and it is working again for non-nordic boards.
```ini
[env]
lib_deps = 
    nrf24/RF24Network @ 1.0.19
    nrf24/RF24Mesh @ 1.1.9

[env:qtpy-esp32]
platform = espressif32
framework = arduino
board = adafruit_qtpy_esp32

```
Unfortunately, I'm too late to retract the RF24Network v1.0.18 release. According to the [PIO docs](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_unpublish.html):
> You can only remove a package version within 72 hours from the published date.

I'm closing this now that it should be resolved with the new releases of nrf2nrf and RF24Network.

closed 2023-06-21 by 2bndy5

shyney7, 2023-06-21:
> Just published RF24Network v1.0.19 and it is working again for non-nordic boards.
>
> ```ini
> [env]
> lib_deps =
>     nrf24/RF24Network @ 1.0.19
>     nrf24/RF24Mesh @ 1.1.9
>
> [env:qtpy-esp32]
> platform = [...]

Thanks for the fast solution to this issue. It works now!

2bndy5, 2023-06-21:
@TMRh20 I was also (finally) able to get the PIO CI working again. Apparently, all named deps of a lib are installed to global storage (`pio pkg install -g`) despite any platform-specificity for said lib deps. This means that installing RF24Network via
```shell
pio pkg install -g nrf24/RF24Network
```
will install nrf_to_nrf if there is no platformio.ini file to declare the project's `platform`. For projects using global storage, the recommended CI approach is to skip lib dependencies and install them manually.
```shell
# note the order matters here
pio pkg install -g -f --skip-dependencies -l nrf24/RF24Mesh
pio pkg install -g -f --skip-dependencies -l nrf24/RF24Network
pio pkg install -g -f --skip-dependencies -l nrf24/RF24
```
> **Warning**
> Global storage is officially discouraged by PIO. Probably because `pio` needs additional info from platformio.ini (like `platform`).
>
> Our CI is a special case because we don't maintain a platformio.ini file for building the Arduino examples. Instead we specify the board/platform via `pio ci ...` to build our examples.