nRF24/RF24Network #213: radio.setDataRate( does not seem to be setting anything

FlailAway, 2023-07-03:
RF24Network 2.0.0, Arduino IDE 1.8.19, Linux Mint+MATE 20.1

This is my common setup for all Nodes. But, at the bottom where I print back the **actual** settings, "DataRate is zero!

What gives? Me or Library? :)

This is in "Setup"
    radioSetup(NODE_ME, PA_LEVEL_HIGH, SPEED_1MB);

This is the function.
  ///////////////////////////////////////////
  void radioSetup(const uint16_t aNode, uint8_t paLevel, uint8_t dataRate) {
    radio.begin();
    radio.setChannel(RADIO_CHANNEL);  // currently 110
//      radio.setAutoAck(true);
//      radio.enableAckPayload();
//      radio.enableDynamicPayloads();
//      radio.setRetries(10,15);      // Max is 15. 0 means 250us, 15 means 4000us.
    switch(paLevel) {
      case 1:
        radio.setPALevel(RF24_PA_MIN);
      break;
      case 2:
        radio.setPALevel(RF24_PA_LOW);
      break;
      case 3:
        radio.setPALevel(RF24_PA_HIGH);
      break;
      case 4:
        radio.setPALevel(RF24_PA_MAX);
      break;
      default:
        radio.setPALevel(RF24_PA_LOW);
      break;
    }
    switch(dataRate) {
      case 1:
        radio.setDataRate(RF24_250KBPS);
      break;
      case 2:
        radio.setDataRate(RF24_1MBPS);
      break;
      case 3:
        radio.setDataRate(RF24_2MBPS);
      break;
      default:
        radio.setDataRate(RF24_1MBPS);
      break;
    }
    network.begin(aNode);
    Serial.print(F("Pin CE "));
    Serial.print(MY_RADIO_CE);
    Serial.print(F(", Pin CSN "));
    Serial.print(MY_RADIO_CSN);
    Serial.print(F(", Channel "));
    Serial.print(radio.getChannel());
    Serial.print(F(", Data Rate "));
    Serial.print(radio.getDataRate());
    Serial.print(F(", PA-Speed "));
    Serial.print(radio.getPALevel());
    Serial.println(F(">"));
  }

2bndy5, 2023-07-03:
Are you using an ebyte module? They reportedly don't allow changing the data rate, but that seems to vary on the ebyte module's model.

2bndy5, 2023-07-03:
> dots 'cos git stripped spaces AND indentation with Edit

Github comments use [MarkDown syntax](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax). What you're looking for (but the site UI has no button for) is called [fenced code blocks](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#fenced-code-blocks).

FlailAway, 2023-07-04:
Thanks for the quick reply, 

I am using the standard Arduino-IDE and all that it encompasses as Stock. Nodes are all basic Handling things like DHT etc no GPS required, I can see all the Node-sites from here. :)

Which Data-Rate is it using then? I like the 250KB as I have long-ish distance between some nodes and that has proved to be 100% reliable and fast enough for this project. Nodes only report in at 1-minute or 2-minute intervals. 

Thanks for the  "fenced code blocks" pointer.

2bndy5, 2023-07-04:
> the Zero is the Enum of the first item that is Zero

I had a feeling you had missed that but wasn't sure. Does this mean the issue is resolved?

Also, have you found a resolution for your other issue (nRF24/RF24#909)? I ask because github isn't reddit; we like to close issues if somehow resolved or become very old/stale.

---
As for the MD syntax, I ~don't~ rarely use the UI buttons. I have become accustomed to writing directly in MarkDown 😜. I have found that the UI in the GitHub mobile app is better than the site's UI

FlailAway, 2023-07-10:
>> BTW, I have found that the UI in the GitHub mobile app 

{grin} You City folk get it easy. We have no cell coverage out here in the boonies where we farm to feed the City folk.

I will close this issue due to operator error

The other issue I **think** is a Octal-Node-numbering thing. I will jump over to that thread to explain.