nRF24/RF24Network #214: Multichannel networks - Separate channels for different levels

TMRh20, 2023-09-23:
I realized that RF24Network could support a multichannel network, with each "level" using a different channel. ie: node 00 uses channel 90, nodes 0x use 95, 0xx use 100 and so on.

In testing this out, the code changes seem to be minimal, and it should improve performance for networks with lots of nodes. Its still in preliminary testing, so no PR yet, but thought I would put this idea out there.

This means each network would take up quite a bit of radio spectrum as well.

2bndy5, 2023-09-23:
This is intriguing! I'm curious to see the code, namely if you wait after the switching channels. Can you create a branch or is it still too early?

Would there be a way to blacklist channels for envs that have specifically noisy channels?

TMRh20, 2023-09-23:
Added a branch. There is still a bit of code lacking when doing a direct write to a node, it needs to figure out the node level and set the channel accordingly, but everything else should be functional.

I don't currently do any waiting after setting the channel, but I need to test this out further etc.

There would be a way to blacklist channels, right now it just uses a base channel and adds 5 for each level.

TMRh20, 2023-09-24:
I'm having no luck with this, going to abandon the idea for now. It seems like any sort of channel hopping on these devices is doomed lol.

closed 2023-09-24 by TMRh20

2bndy5, 2023-09-24:
Yeah, channel hopping is typically done with time slots/intervals. After a couple hours of theorizing this idea, I couldn't imagine it working with the same/current reliability.

If channel hopping is the underlying goal, then 
1. `network.update()` would have to encompass managing the channel as time elapses.
2. RF24Network would need a new network-reserved message type to handle clock syncing (probably as a multicast).

This added complexity would inherently cost in speed and/or reliability though.