nRF24/RF24Network #222: [Q] How to create multiple networks in the same area/range?

xdxita, 2024-03-15:
Hello! Thanks for this great library. I'm waiting for my nRF24 modules to arrive, in the meantime I'm getting used to the code.
From what I've read so far, it wouldn't be possible, but..

Is it possible to have 2 or more networks each one with it's master and nodes both networks are in range of each other.
How to prevent an unwanted node to join the network? Are any "security" features available?

Thanks!

2bndy5, 2024-03-15:
> How to prevent an unwanted node to join the network? 

Use a different channel for each network. Meaning,
- all nodes (+ master) in network A use channel 90
- all nodes (+ master) in network A use channel 100

See [`RF24::setChannel()`](https://nrf24.github.io/RF24/classRF24.html#a5e6e5a5f6c85d2638381cab2c0f3702e)

> Are any "security" features available?

No, it was designed like that on purpose. You are free to implement your own security measures, but we don't support such endeavors.

TMRh20, 2024-06-14:
>  Are any "security" features available?

Technically yes, with NRF52x devices, encryption and authentication is supported and built into the library. With NRF24, no.

closed 2024-08-02 by xdxita