nRF24/RF24Network #233: Network Default Address (clarification please)

luudee, 2024-06-29:
Hi !

First of all I would like to say you are doing an absolutely excellent job developing this library !
I am working on a large project that will deploy thousands of nodes over a wide area !

I understand the network topology, and you state in the online description that 05555 would
be the highest address achievable, supporting 3,125  nodes. However, it appears you are
using 04444 as the network "default address".  I am not sure I would really use the 3,125
nodes, but I can see how in my case 04444 could belong to a valid node.

Is it ok to change this value to 07777 as there would be no real node with that address ?
I am a bit confused what the purpose if this address is anyway. Could you be so kind and
comment on this ?

Also, may I humbly suggest to not hard code values like 04444 but rather use defines for
this ? It would make future changes and upgrades so much easier. :)

Many Thanks,
rudi

2bndy5, 2024-06-29:
> Is it ok to change this value to 07777

I was pondering this again... With the recent addition of nRF5x support (which can use 8 pipes), I think the sentinel value 07777 becomes a valid address when `NUM_PIPES` is set to 8. Ultimately, I'm hesitant to change `NETWORK_DEFAULT_ADDRESS` value at all due to backward compatibility issues because every node in the network would need to be updated.

If we do change this (to properly allow a mesh child node to actually use the assigned address 04444), then it would have to be something that is not a valid address under any situation.

Given that there are 5 network levels and level 0 is master, the addresses rarely use the 3 MSBits in a signed 16 bit variable (excluding the signed bit). I believe these top 3 bits could be compared with sentinel masks to quickly identify when the address is being used for a special purpose.
- `0b001 << 12`: multicast to level 0
- `0b010 << 12`: multicast to level 1
- `0b100 << 12`: multicast to level 2
- `0b110 << 12`: multicast to level 3
- `0b101 << 12`: multicast to level 4
- `0b011 << 12`: `MESH_BLANK_ID` (see [`RF24Mesh::getNodeID()`](https://nrf24.github.io/RF24Mesh/classESBMesh.html#a070bfa1f5dea90e417e9131407940823))
- `0b111 << 12`: `NETWORK_DEFAULT_ADDRESS` (& synonymous with `MESH_DEFAULT_ADDRESS`)

Notice the above mask examples do not use the MSBit (`1 << 15`). This is because `RF24Mesh::getAddress()` & `RF24Mesh::getNodeID()` currently can return a negative number (-1 and -2).

TMRh20, 2024-06-29:
> If we do change this (to properly allow a mesh child node to actually use the assigned address 04444), then it would have to be something that is not a valid address under any situation.

And I'm not sure that would work, because it would need a valid address of some sort to function with RF24Mesh.

> I believe these top 3 bits could be compared with sentinel masks to quickly identify when the address is being used for a special purpose.

Interesting, would that make things more or less confusing? :smile:

2bndy5, 2024-06-29:
All of this conjecture is aimed at allowing mesh child nodes to be assigned 04444.

I think it could be made _just as_ confusing if the code is written "adequately." 😝 

> because it would need a valid address of some sort to function with RF24Mesh

Yeah, I glossed over this. I think that valid addresses could still be used, but the identity of nodes without an assigned address would be determined by masking the address...
```c
#define NETWORK_DEFAULT_ADDRESS 070000 // 0b111 << 12
if (header.from & NETWORK_DEFAULT_ADDRESS == NETWORK_DEFAULT_ADDRESS) {
    // node is trying to connect
}
```

2bndy5, 2024-06-30:
Looking at how my masking idea would be implemented, I see how it would compromise multicasting. I really thought I was on to something, but to use a mask instead of a sentinel would need an overhaul of address-to-pipe handling... Nevermind. I guess child nodes can never actually use address 04444. 😞

TMRh20, 2024-06-30:
Well ideas are like that sometimes, you gotta poke around a little to see if they work.