nRF24/RF24Network #242: NRF24 Network: network.available() returns true in every loop iteration when sending frequent messages from the master node

mirhamza708, 2025-01-04:
I'm experiencing an issue where the network.available() function in the receiveCommand method returns true in each loop iteration, even when the master node is not sending any data. This issue occurs only when a large number of messages are sent quickly from the master node (with intervals reduced to around 100ms-200ms). When a delay of approximately 500ms is introduced between messages, the issue does not occur, and the function behaves as expected.

**Steps to Reproduce:**

1. Set up the NRF24Handler on a node to receive data.
2. Use the master node to send frequent messages with intervals of around 100ms-200ms and then after 5-10 seconds stop sending new data.
3. Observe the behavior in the receiveCommand method where network.available() returns true in every iteration of the loop, even when no new data is received. The while loop is exited but in the next iteration it returns true again.
4. Reducing the interval between messages (e.g., to 100ms) triggers this issue.
5. Introducing a delay of 500ms between messages resolves the issue.
6. Expected Behavior: network.available() should return true only when new data is available for reading, not in every iteration of the loop.

**Actual Behavior:** When a lot of messages are sent rapidly from the master node, network.available() returns true in every iteration of the loop, even when no new data has been received.

nrf24Handler class methods
```

void NRF24Handler::begin(uint16_t nodeAddress, uint8_t network_channel) {
    setChannel(network_channel);
    setNodeAddress(nodeAddress);
    
    if (radio.begin()) {
        network.begin(network_channel, nodeAddress); // 90 is the RF24 channel
        radio.setPALevel(RF24_PA_HIGH);  // Adjust power level as needed
        radio.setDataRate(RF24_1MBPS);  // 1Mbps for better range
        Serial.println("NRF24 Network: Initialized successfully.");
    } else {
        Serial.println("NRF24 Network: Failed to initialize radio hardware. Check connections.");
    }
}

void NRF24Handler::sendData(const uint8_t* buffer, size_t bufferSize) {
    network.update();
    RF24NetworkHeader header(masterAddress, 'D'); // 'D' is the message type
    Serial.print("Size of buffer: ");
    Serial.println(bufferSize);
    
    // Write the buffer to the network
    bool success = network.write(header, buffer, bufferSize);

    if (success) {
        Serial.println("NRF24 Network: Data sent to master node.");
    } else {
        Serial.println("NRF24 Network: Failed to send data.");
    }
}

bool NRF24Handler::receiveCommand(ThermostatData& data) {
    network.update();
    
    while (network.available()) {
        RF24NetworkHeader header;
        uint16_t payloadSize = network.peek(header);     // Use peek() to get the size of the payload
        uint8_t buffer[payloadSize]; // Adjusted to the size of the expected command payload (6 bytes)
        // Read the data into the temporary buffer
        network.read(header, buffer, payloadSize);

        // Check if the message type matches the command type 'C'
        switch (header.type) {
            case 'C': // 'C' is the command message type
                Serial.println("NRF24 Network: Command received from master node.");

                // Parse the buffer into ThermostatData
                assign_if_valid(data.state, buffer[0], {ThermostatState::Off, ThermostatState::On}); // 1st byte: Thermostat state
                assign_if_valid(data.mode, buffer[1], {ThermostatMode::Auto, ThermostatMode::Manual, ThermostatMode::Ventilation}); // 2nd byte: Thermostat mode
                assign_if_valid(data.fanSpeed, buffer[2], {ThermostatFanSpeed::Low, ThermostatFanSpeed::Medium, ThermostatFanSpeed::High}); // 3rd byte: Fan speed
                
                if (buffer[3] >= 16 && buffer[3] <= 32) {
                    data.setTemperature = buffer[3]; // 4th byte: Set temperature
                }

                // Debug print for verification
                Serial.println("Parsed Command:");
                Serial.print("State: ");
                Serial.println(stateToString(data.state));
                Serial.print("Mode: ");
                Serial.println(modeToString(data.mode)); // Convert `ThermostatMode` to string
                Serial.print("Fan Speed: ");
                Serial.println(fanSpeedToString(data.fanSpeed)); // Use fanSpeed for fan speed
                Serial.print("Set Temperature: ");
                Serial.println(data.setTemperature);

                return true;
            case 'S':
                Serial.println("Settings received!");
                switch (buffer[0]) {
                    case 1:
                        Serial.println("Setting 1");
                        break;
                    case 2:
                        Serial.println("Setting 2");
                        break;
                    case 3:
                        Serial.println("Setting 3");
                        break;
                    default:
                        Serial.println("Unknown settings");
                        break;
                }
                break;
        }
    }
    return false;
}
```

my main.cpp
```
NRF24Handler nrfHandler(NRF_CE_PIN, NRF_CSN_PIN);

/**
 * @brief Checks for incoming commands from the master node.
 *
 * If a new command is received, it updates the thermostat state and sends the updated data to the LCD.
 */
void checkForNrfCommands() {
    if (nrfHandler.receiveCommand(thermostat)) {
        Serial.println("NRF24: Thermostat updated with new command.");
        lcd.dwinPageChange(static_cast<uint16_t>(thermostat.state));
        lcd.updateModeIcon(thermostat.mode);
        if (thermostat.mode != ThermostatMode::Auto) {
            lcd.setFanSpeedIcon(static_cast<uint16_t>(thermostat.fanSpeed));
        }
        lcd.updateSetTemperature(thermostat.setTemperature);
    }
}

void setup() {
    Serial.begin(9600);
    Serial.setTimeout(1000);

    delay(2000);

    // Initialize nRF24 Network
    nrfHandler.begin(01, 90);

}

void loop() {
    if (millis() - lastDataSendTime >= 5000) {
        lastDataSendTime = millis();
        sendDataToMasterNode();
    }

    checkForNrfCommands();
}
```

TMRh20, 2025-01-04:
I am unable to recreate any issue using your code.

mirhamza708, 2025-01-04:
Let me create a simple example to share here.

mirhamza708, 2025-01-04:
> I am unable to recreate any issue using your code.

I think I found the problem. Nothing wrong in the library but if the network update is not called often enough then this happens. Can you tell me how often, the frequency, should this update method be called?

TMRh20, 2025-01-04:
`network.update` needs to be called as often as possible in a high traffic environment, typically at a rate equal to or faster than the transmitting device(s) are sending.

Typically, you want to call `network.update()` very regularly, and it will load the data into a buffer. 
The `available()` function can be called at a slower rate, allowing you to process data at a slower rate if necessary.

Keep in mind, the radios will buffer up to 3 payloads, while RF24Network will buffer up to 144 bytes of data by default. The radio will also keep receiving data while the microcontroller is doing other things, as long as its in RX mode.

mirhamza708, 2025-01-04:
But even if this is called at a slower rate the network.available should not return true if there is no payload for this node. If due to any reason example some heavy calculation or some other process delays the network update the network.available should not break.

TMRh20, 2025-01-04:
Nope, I would suggest a potential issue with your code if that is happening. 

Often, if you have buffer overwrites or other memory related issues, it tends to invoke strange behavior that cannot be easily explained.

If you can post some simplified code that demonstrates this behavior, I can take a closer look.

mirhamza708, 2025-01-04:
yeah sure I will share.

mirhamza708, 2025-01-04:
So here is the node side code
```
#include <Arduino.h>
#include <RF24.h>
#include <RF24Network.h>

const uint8_t STATUS_LED = PC13; ///< Pin for nRF24 CSN

HardwareSerial Serial1(USART1);

const uint8_t NRF_CE_PIN = PB0; ///< Pin for nRF24 CE
const uint8_t NRF_CSN_PIN = PB1; ///< Pin for nRF24 CSN

unsigned long lastDataSendTime = 0; ///< Last time data was sent to the master node
RF24 radio(NRF_CE_PIN, NRF_CSN_PIN);
RF24Network network(radio);

void sendData(const uint8_t* buffer, size_t bufferSize) {
    RF24NetworkHeader header(00, 'D'); // 'D' is the message type
    Serial.print("Size of buffer: ");
    Serial.println(bufferSize);
    
    // Write the buffer to the network
    bool success = network.write(header, buffer, bufferSize);

    if (success) {
        Serial.println("NRF24 Network: Data sent to master node.");
    } else {
        Serial.println("NRF24 Network: Failed to send data.");
    }
}

bool receiveCommand(void) {
    network.update();
    
    while (network.available()) {
        RF24NetworkHeader header;
        uint16_t payloadSize = network.peek(header);     // Use peek() to get the size of the payload
        uint8_t buffer[payloadSize]; // Adjusted to the size of the expected command payload (6 bytes)
        // Read the data into the temporary buffer
        network.read(header, buffer, payloadSize);

        // Check if the message type matches the command type 'C'
        switch (header.type) {
            case 'C': // 'C' is the command message type
                Serial.println("NRF24 Network: Command received from master node.");
                return true;
            case 'S':
                Serial.println("NRF24 Network: Settings received from master node.");
                break;
        }
    }
    return false;
}

void setup() {
    Serial.begin(57600);
    Serial.setTimeout(1000);
    pinMode(STATUS_LED, OUTPUT);
    delay(2000);

    if (radio.begin()) {
        network.begin(90, 01); // 90 is the RF24 channel
        radio.setPALevel(RF24_PA_HIGH);  // Adjust power level as needed
        radio.setDataRate(RF24_1MBPS);  // 1Mbps for better range
        Serial.println("NRF24 Network: Initialized successfully.");
    } else {
        Serial.println("NRF24 Network: Failed to initialize radio hardware. Check connections.");
    }
}
[... 195 lines cut ...]
backup_count = 3

# Configure logging
logger = logging.getLogger("gateway_logger")
logger.setLevel(logging.INFO)

# Handlers
console_handler = logging.StreamHandler()  # Terminal
console_handler.setFormatter(logging.Formatter("%(asctime)s - %(levelname)s - %(message)s"))

file_handler = RotatingFileHandler(log_file, maxBytes=max_log_size, backupCount=backup_count)
file_handler.setFormatter(logging.Formatter("%(asctime)s - %(levelname)s - %(message)s"))

# Apply the same formatter to InMemoryLogHandler
log_formatter = logging.Formatter("%(asctime)s - %(levelname)s - %(message)s")
in_memory_handler.setFormatter(log_formatter)

# Add handlers to logger
logger.addHandler(console_handler)
logger.addHandler(file_handler)
logger.addHandler(in_memory_handler)
```

Maybe you can run either on raspberry pi or orange pi zero 3 lts (if you have) but I think in master there shouldn't be any issue
sorry I couldn't strip it down any more that's the minimum code.

TMRh20, 2025-01-04:
I still can't recreate using your Arduino example and can't run your python example for now. 

1. What Arduino board are you running?
2. Are you able to test/recreate the issue with two Arduino boards? I modified the code below to send to your example. How about if you use two Arduinos and the following code or a modified version:

Send a 's' over serial to stop sending and 'g' to resume.

```
#include <SPI.h>
#include <RF24.h>
#include <RF24Network.h>

RF24 radio(7, 8);  // nRF24L01(+) radio attached using Getting Started board

RF24Network network(radio);  // Network uses that radio

const uint16_t this_node = 00;   // Address of our node in Octal format
const uint16_t other_node = 01;  // Address of the other node in Octal format

const unsigned long interval = 0;  // How often (in ms) to send 'hello world' to the other unit

unsigned long last_sent;     // When did we last send?
unsigned long packets_sent;  // How many have we sent already

struct payload_t {  // Structure of our payload
  unsigned long ms;
  unsigned long counter;
};

void setup(void) {
  Serial.begin(115200);
  while (!Serial) {
    // some boards need this because of native USB capability
  }
  Serial.println(F("RF24Network/examples/helloworld_tx/"));

  if (!radio.begin()) {
    Serial.println(F("Radio hardware not responding!"));
    while (1) {
      // hold in infinite loop
    }
  }
  radio.setChannel(90);
  network.begin(/*node address*/ this_node);
}

bool sending = 1;

void loop() {

  if(Serial.available()){
    char c = Serial.read();
    if(c == 'g'){
      sending = 1;
    }else
    if(c == 's'){
      sending = 0;
    }
  }

  network.update();  // Check the network regularly

  unsigned long now = millis();

  // If it's time to send a message, send it!
  if (now - last_sent >= interval) {
    last_sent = now;
    if(sending){
    Serial.print(F("Sending... "));
    payload_t payload = { millis(), packets_sent++ };
    RF24NetworkHeader header(/*to node*/ other_node);
    bool ok = network.write(header, &payload, sizeof(payload));
    Serial.println(ok ? F("ok.") : F("failed."));
    }
  }

  while (network.available()) {  // Is there anything ready for us?

    RF24NetworkHeader header;  // If so, grab it and print it out
    
    network.read(header, 0, 0);
    Serial.println(F("Received packet: counter="));

  }

}

```

TMRh20, 2025-01-04:
Maybe we should call in the expert: @2bndy5 any thoughts?

2bndy5, 2025-01-04:
```
      switch (header.type) {
            case 'C': // 'C' is the command message type
                Serial.println("NRF24 Network: Command received from master node.");

                return true;
            case 'S':
                Serial.println("Settings received!");

                break;
        }
```
When you hit this while `available()` remains true, does it print anything?

---

It's been a while since I went over the internal pointer that manages the network "queue" for non-linux platforms. The user code posted here seems right at a glance.

mirhamza708, 2025-01-04:
Yeah it goes to the case 'C'

mirhamza708, 2025-01-04:
> I still can't recreate using your Arduino example and can't run your python example for now.
> 
> 1. What Arduino board are you running?
> 2. Are you able to test/recreate the issue with two Arduino boards? I modified the code below to send to your example. How about if you use two Arduinos and the following code or a modified version:
> 
> Send a 's' over serial to stop sending and 'g' to resume.
> 
> ```c++
> #include <SPI.h>
> #include <RF24.h>
> #include <RF24Network.h>
> 
> RF24 radio(7, 8);  // nRF24L01(+) radio attached using Getting Started board
> 
> RF24Network network(radio);  // Network uses that radio
> 
> const uint16_t this_node = 00;   // Address of our node in Octal format
> const uint16_t other_node = 01;  // Address of the other node in Octal format
> 
> const unsigned long interval = 0;  // How often (in ms) to send 'hello world' to the other unit
> 
> unsigned long last_sent;     // When did we last send?
> unsigned long packets_sent;  // How many have we sent already
> 
> 
> struct payload_t {  // Structure of our payload
>   unsigned long ms;
>   unsigned long counter;
> };
> 
> void setup(void) {
>   Serial.begin(115200);
>   while (!Serial) {
>     // some boards need this because of native USB capability
>   }
>   Serial.println(F("RF24Network/examples/helloworld_tx/"));
> 
>   if (!radio.begin()) {
>     Serial.println(F("Radio hardware not responding!"));
>     while (1) {
>       // hold in infinite loop
>     }
>   }
>   radio.setChannel(90);
>   network.begin(/*node address*/ this_node);
> }
> 
> bool sending = 1;
> 
> void loop() {
> 
>   if(Serial.available()){
>     char c = Serial.read();
>     if(c == 'g'){
>       sending = 1;
>     }else
>     if(c == 's'){
>       sending = 0;
>     }
>   }
> 
> 
>   network.update();  // Check the network regularly
> 
>   unsigned long now = millis();
> 
>   // If it's time to send a message, send it!
>   if (now - last_sent >= interval) {
>     last_sent = now;
>     if(sending){
>     Serial.print(F("Sending... "));
>     payload_t payload = { millis(), packets_sent++ };
>     RF24NetworkHeader header(/*to node*/ other_node);
>     bool ok = network.write(header, &payload, sizeof(payload));
>     Serial.println(ok ? F("ok.") : F("failed."));
>     }
>   }
> 
>   while (network.available()) {  // Is there anything ready for us?
> 
>     RF24NetworkHeader header;  // If so, grab it and print it out
>     
>     network.read(header, 0, 0);
>     Serial.println(F("Received packet: counter="));
> 
>   }
> 
> }
> ```
> 
> and in the receiver example I modified mainly the below:
> 
> ```c++
>         switch (header.type) {
>             case 'C': // 'C' is the command message type
>                 Serial.println("NRF24 Network: Command received from master node.");
>                 return true;
>             case 'S':
>                 Serial.println("NRF24 Network: Settings received from master node.");
>                 break;
>             default: Serial.println("xD"); break;
>         }
> ```

I will do this

mirhamza708, 2025-01-04:
> I will do this

Works fine with an Arduino uno as master

mirhamza708, 2025-01-04:
[image]

when the node is in delay then it just fails to transmit. but after stoppin master the node works fine

[image]

TMRh20, 2025-01-04:
Can you confirm that powering off the Orange Pi **AND** disconnecting the radio from the Orange Pi results in the same behavior with your original code?

mirhamza708, 2025-01-04:
> Can you confirm that powering off the Orange Pi **AND** disconnecting the radio from the Orange Pi results in the same behavior with your original code?

working on it

2bndy5, 2025-01-04:
I suspected a hardware problem from the start, but it is strange to see it in the network layer. Usually hardware errors annoyingly surface when using the core layer (RF24 lib). **Have we checked behavior in that layer first?** The payloads aren't big enough to need fragmentation, so recreating this with only RF24 layer shouldn't be too hard.

If it is not a hardware error, then some kind of memory leak might be preventing `available()` from returning accurately. The internal pointer doesn't seem to be misused in a way that would cause this problem.

TMRh20, 2025-01-07:
@mirhamza708 Any luck or progress on this issue?

mirhamza708, 2025-01-07:
> Can you confirm that powering off the Orange Pi **AND** disconnecting the radio from the Orange Pi results in the same behavior with your original code?

So I completely removed the nrf24 from master and the node is still receiving messages this is the code of the node 
```
#include <Arduino.h>
#include <RF24.h>
#include <RF24Network.h>

const uint8_t STATUS_LED = PC13; ///< Pin for nRF24 CSN

HardwareSerial Serial1(USART1);

const uint8_t NRF_CE_PIN = PB0; ///< Pin for nRF24 CE
const uint8_t NRF_CSN_PIN = PB1; ///< Pin for nRF24 CSN

unsigned long lastDataSendTime = 0; ///< Last time data was sent to the master node
RF24 radio(NRF_CE_PIN, NRF_CSN_PIN);
RF24Network network(radio);

void sendData(const uint8_t* buffer, size_t bufferSize) {
    RF24NetworkHeader header(00, 'D'); // 'D' is the message type
    Serial.print("Size of buffer: ");
    Serial.println(bufferSize);
    
    // Write the buffer to the network
    bool success = network.write(header, buffer, bufferSize);

    if (success) {
        Serial.println("NRF24 Network: Data sent to master node.");
    } else {
        Serial.println("NRF24 Network: Failed to send data.");
    }
}

bool receiveCommand(void) {
    network.update();
    
    while (network.available()) {
        RF24NetworkHeader header;
        uint16_t payloadSize = network.peek(header);     // Use peek() to get the size of the payload
        uint8_t buffer[payloadSize]; // Adjusted to the size of the expected command payload (6 bytes)
        // Read the data into the temporary buffer
        network.read(header, buffer, payloadSize);

        // Check if the message type matches the command type 'C'
        switch (header.type) {
            case 'C': // 'C' is the command message type
                Serial.println("NRF24 Network: Command received from master node.");
                return true;
            case 'S':
                Serial.println("NRF24 Network: Settings received from master node.");
                break;
            default: Serial.println("xD"); break;
        }
    }
    return false;
}

void setup() {
    Serial.begin(57600);
    Serial.setTimeout(1000);
    pinMode(STATUS_LED, OUTPUT);
    delay(2000);

    if (radio.begin()) {
        network.begin(90, 01); // 90 is the RF24 channel
        radio.setPALevel(RF24_PA_LOW);  // Adjust power level as needed
        radio.setDataRate(RF24_1MBPS);  // 1Mbps for better range
        Serial.println("NRF24 Network: Initialized successfully.");
    } else {
        Serial.println("NRF24 Network: Failed to initialize radio hardware. Check connections.");
    }
}

void loop() {

    if (millis() - lastDataSendTime >= 5000) {
        lastDataSendTime = millis();
        uint8_t buffer[18];  // Increase the buffer size to 17 to accommodate roomTemperature
        buffer[0] = 1;
        uint64_t some_data = 0x1234567812345678;
        memcpy(&buffer[1], &some_data, sizeof(some_data));

        buffer[9] = 12;
        buffer[10] = 34;
        buffer[11] = 2;
        buffer[12] = 2;
        buffer[13] = 2;
        buffer[14] = 2;
        buffer[15] = 22;
        buffer[16] = 22;
        buffer[17] = 2;

        sendData(buffer, sizeof(buffer));
    }

    receiveCommand();
    delay(500); // after adding this delay the network.available keeps returning true
}
```

TMRh20, 2025-01-07:
I was able to recreate this behavior between Arduino Due and Arduino Nano with the following as TX:

```
/**

#include <SPI.h>
#include <RF24.h>
#include <RF24Network.h>

RF24 radio(7, 8);  // nRF24L01(+) radio attached using Getting Started board

RF24Network network(radio);  // Network uses that radio

const uint16_t this_node = 00;   // Address of our node in Octal format
const uint16_t other_node = 01;  // Address of the other node in Octal format

const unsigned long interval = 50;  // How often (in ms) to send 'hello world' to the other unit

unsigned long last_sent;     // When did we last send?
unsigned long packets_sent;  // How many have we sent already

//data1 = 1
//                data2 = 2
//                data3 = 2
//                data4 = 22
//                # Prepare the payload
//                payload = struct.pack("<BBBB", data1, data2, data3, data4)

struct payload_t {  // Structure of our payload
  uint8_t data1 = 1;
  uint8_t data2 = 2;
  uint8_t data3 = 2;
  uint8_t data4 = 22;
};

void setup(void) {
  Serial.begin(115200);
  while (!Serial) {
    // some boards need this because of native USB capability
  }
  Serial.println(F("RF24Network/examples/helloworld_tx/"));

  if (!radio.begin()) {
    Serial.println(F("Radio hardware not responding!"));
    while (1) {
      // hold in infinite loop
    }
  }
  radio.setChannel(90);
  network.begin(/*node address*/ this_node);
}

bool send = 1; 

void loop() {
   
   if(Serial.available()){
     char c = Serial.read();
     if(c == 'g'){
      send = 1;
     }else
     if(c == 's'){
      send = 0;
     }
   }
   

  network.update();  // Check the network regularly

  unsigned long now = millis();

  // If it's time to send a message, send it!
  if (now - last_sent >= interval && send) {
    last_sent = now;

    Serial.print(F("Sending... "));
    //payload_t payload = { millis(), packets_sent++ };
    payload_t payload;
    RF24NetworkHeader header(/*to node*/ other_node, 'C');
    bool ok = network.write(header, &payload, sizeof(payload));
    Serial.println(ok ? F("ok.") : F("failed."));
  }
}
```

Houston, we've had a problem...

This is going to take some debugging to fix.

mirhamza708, 2025-01-07:
> https://github.com/nRF24/RF24Network/blob/master/RF24Network.cpp#L518
> 
> I think this should be `if (message_size + (next_frame - frame_queue) <= MAX_PAYLOAD_SIZE) { `
> 
> Are you able to test this change out @mirhamza708 ?

Yeah just check it resolves the issue, but need to test more.

mirhamza708, 2025-01-07:
> https://github.com/nRF24/RF24Network/blob/master/RF24Network.cpp#L518
> 
> I think this should be `if (message_size + (next_frame - frame_queue) <= MAX_PAYLOAD_SIZE) { `
> 
> Are you able to test this change out @mirhamza708 ?

Works fine with my setup.

mirhamza708, 2025-01-07:
> ```
> 2025-01-07 12:37:13,087 - INFO - Message sent to node 1.
> 2025-01-07 12:37:13,190 - INFO - Message sent to node 1.
> 2025-01-07 12:37:13,296 - INFO - Message sent to node 1.
> 2025-01-07 12:37:13,431 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:13,536 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:13,608 - INFO - Message sent to node 1.
> 2025-01-07 12:37:13,715 - INFO - Message sent to node 1.
> 2025-01-07 12:37:13,818 - INFO - Message sent to node 1.
> 2025-01-07 12:37:13,952 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:14,057 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:14,130 - INFO - Message sent to node 1.
> 2025-01-07 12:37:14,234 - INFO - Message sent to node 1.
> 2025-01-07 12:37:14,337 - INFO - Message sent to node 1.
> 2025-01-07 12:37:14,472 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:14,576 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:14,649 - INFO - Message sent to node 1.
> 2025-01-07 12:37:14,759 - INFO - Message sent to node 1.
> 2025-01-07 12:37:14,854 - INFO - Message sent to node 1.
> 2025-01-07 12:37:14,991 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:15,097 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:15,173 - INFO - Message sent to node 1.
> 2025-01-07 12:37:15,278 - INFO - Message sent to node 1.
> 2025-01-07 12:37:15,382 - INFO - Message sent to node 1.
> 2025-01-07 12:37:15,517 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:15,611 - INFO - Message sent to node 1.
> 2025-01-07 12:37:15,696 - INFO - Message sent to node 1.
> 2025-01-07 12:37:15,799 - INFO - Message sent to node 1.
> 2025-01-07 12:37:15,933 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:16,038 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:16,111 - INFO - Message sent to node 1.
> 2025-01-07 12:37:16,215 - INFO - Message sent to node 1.
> 2025-01-07 12:37:16,318 - INFO - Message sent to node 1.
> 2025-01-07 12:37:16,453 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:16,557 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:16,630 - INFO - Message sent to node 1.
> 2025-01-07 12:37:16,734 - INFO - Message sent to node 1.
> 2025-01-07 12:37:16,843 - INFO - Message sent to node 1.
> 2025-01-07 12:37:16,977 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:17,082 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:17,160 - INFO - Message sent to node 1.
> 2025-01-07 12:37:17,269 - INFO - Message sent to node 1.
> 2025-01-07 12:37:17,375 - INFO - Message sent to node 1.
> 2025-01-07 12:37:17,512 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:17,606 - INFO - Message sent to node 1.
> 2025-01-07 12:37:17,692 - INFO - Message sent to node 1.
> 2025-01-07 12:37:17,797 - INFO - Message sent to node 1.
> 2025-01-07 12:37:17,933 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:18,039 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:18,116 - INFO - Message sent to node 1.
> 2025-01-07 12:37:18,225 - INFO - Message sent to node 1.
> 2025-01-07 12:37:18,335 - INFO - Message sent to node 1.
> 2025-01-07 12:37:18,471 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:18,578 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:18,653 - INFO - Message sent to node 1.
> 2025-01-07 12:37:18,758 - INFO - Message sent to node 1.
> 2025-01-07 12:37:18,864 - INFO - Message sent to node 1.
> 2025-01-07 12:37:18,999 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:19,103 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:19,176 - INFO - Message sent to node 1.
> 2025-01-07 12:37:19,282 - INFO - Message sent to node 1.
> 2025-01-07 12:37:19,389 - INFO - Message sent to node 1.
> 2025-01-07 12:37:19,524 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:19,628 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:19,703 - INFO - Message sent to node 1.
> 2025-01-07 12:37:19,809 - INFO - Message sent to node 1.
> 2025-01-07 12:37:19,914 - INFO - Message sent to node 1.
> 2025-01-07 12:37:20,048 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:20,139 - INFO - Message sent to node 1.
> 2025-01-07 12:37:20,225 - INFO - Message sent to node 1.
[... 118 lines cut ...]
> 2025-01-07 12:37:32,727 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:32,832 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:32,937 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,043 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,148 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,252 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,357 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,462 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,566 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,671 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,776 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,880 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:33,985 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:34,089 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:34,194 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:34,299 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:34,403 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:34,508 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:34,612 - ERROR - Failed to send message to node 1.
> 2025-01-07 12:37:34,717 - ERROR - Failed to send message to node 1.
> ```
> 
> Somehow after sending these messages the master node keeps failing. Restarting the app resolves the issue.

This is not after the change this was there even before the change.

TMRh20, 2025-01-07:
Lol more issues?

Can you include something like `if (radio.failureDetected > 0 || radio.getDataRate() != RF24_1MBPS)` but in python, to see if you are encountering radio errors on the OrangePi side of things?

mirhamza708, 2025-01-07:
```
        if self.radio.failureDetected > 0 or self.radio.getDataRate() != RF24_1MBPS:
            return True
        return False
```
I don't know this always returns true is this correct python implementation?

2bndy5, 2025-01-07:
```
2025-01-07 12:37:14,649 - INFO - Message sent to node 1.
2025-01-07 12:37:14,759 - INFO - Message sent to node 1.
2025-01-07 12:37:14,854 - INFO - Message sent to node 1.
2025-01-07 12:37:14,991 - ERROR - Failed to send message to node 1.
2025-01-07 12:37:15,097 - ERROR - Failed to send message to node 1.
```

This looks to me like the receiver's RX FIFO is getting full and it takes the time of 2 failed transmissions to free up it's RX FIFO.

TMRh20, 2025-01-07:
That is odd. Maybe try just  `if self.radio.failureDetected > 0`  ?

I would like to somehow isolate the problem to either the Orange Pi or the STM32. It works for me, so not quite sure how to address this problem yet.

mirhamza708, 2025-01-08:
This always returned 1 for failureDetected. The self.radio.getDataRate() reads 1MBPS.
```
    def radio_failure_check(self):
        """
        Check if there were any radio failures
        """
        # if self.radio.failureDetected > 0:
        #    return 1
        if self.radio.getDataRate() != RF24_1MBPS:
            return 2
        return 0
```

mirhamza708, 2025-01-08:
> This is not after the change this was there even before the change.

there was a mistake in this from my side, it works now. The only issue is that the ```self.radio.failureDetected``` always return greater than zero.

TMRh20, 2025-01-08:
How about using something like the following, again though in python:

```
if(radio.failureDetected) {
  radio.begin();
  radio.setChannel(90);
  network.begin();
  radio.failureDetected = 0;
  Serial.println("Failure Detected");
}
```

Wondering if you really are getting failures, or if there is something wrong in the code. Will this print "Failure Detected" in rapid succession, or will it be intermittent or go away entirely?

TMRh20, 2025-01-08:
Hmm, I think now this is just because the failureDetected var is not initialized in the library at runtime. Something we should correct.

It uses a simple timer to detect timeouts in SPI communication. So if the radio is unresponsive via SPI, the failureDetected will return true. Sometimes SPI does give data, but it is garbage data, so I typically check in internal variable like the dataRate as well, in attempts to correct all SPI/radio errors.

mirhamza708, 2025-01-08:
I would like to correct/contribute to it but I don't understand how the python code is written I see cpp files in the pyrf24 repo all around so I guess I will have to wait for your update. Thanks guys :thumbsup:

TMRh20, 2025-01-08:
Basically, to fix temporarily you can initialize failureDetected to 0 like you are doing. It should still work afterwards. I will go ahead an open an issue on your behalf to fix it.

Thanks alot for identifying these issues! It helps to have really good info and communication when trying to figure out what exactly is going on.

closed 2025-01-08 by TMRh20