nRF24/RF24Network #246: Don't set OK variable in write_to_pipe() function if writeFast() fails

TMRh20, 2025-04-27:
Found this just now, looking at required changes for RF24 bug.

https://github.com/nRF24/RF24Network/blob/380efb83c9ee8d0455279926fd9a6944dff5f437/RF24Network.cpp#L802-L807

I propose changing it to:

```cpp
    header.type = type;
    if (networkFlags & FLAG_FAST_FRAG) {
        radio.startListening();
    }
```

https://github.com/nRF24/RF24Network/blob/380efb83c9ee8d0455279926fd9a6944dff5f437/RF24Network.cpp#L1028-L1036

I propose changing it to:
```cpp
    ok = radio.writeFast(frame_buffer, frame_size, 0);

    if (!ok) {
       radio.txStandBy(txTimeout);      
       radio.setAutoAck(0, 0);
    }else
    if ( (!(networkFlags & FLAG_FAST_FRAG)) || frame_buffer[6] == NETWORK_LAST_FRAGMENT ) {
        ok = radio.txStandBy(txTimeout);
        radio.setAutoAck(0, 0);
    }
```
There are really two changes here:
1. Don't set OK if it already failed. Just try to complete TX of the packets in the FIFO. The OK var SHOULD return false.
2. If FAST_FRAG flag isn't set, (single send) & `writeFast()` didn't fail, it should set the OK var & go into txStandby, but it should also do this on the last fragment.

2bndy5, 2025-04-27:
1. Makes sense because in `write_to_pipe()` https://github.com/nRF24/RF24Network/blob/380efb83c9ee8d0455279926fd9a6944dff5f437/RF24Network.cpp#L1028-L1036
2. This is where you lose me. You want to change the reason proposal 1 makes sense. Sounds like an oversight where success of last fragment wasn't properly reported. But I'm having trouble with the numerous `*write*()` definitions here.

2bndy5, 2025-04-27:
Oh I see what you did in proposal 2. You changed around the logic and added a check for the last fragment. Ok, that makes sense now too.

PR #248 (2025-04-27): Fix write to pipe
The main purpose of this fix is to return failure on a single write failure.

Previous behavior would "skip" the current payload and return the status of previously sent packets in the TX FIFO.

Closes #246
closed 2025-04-27 by TMRh20