nRF24/RF24Network #247: Re-send lost fragments

TMRh20, 2025-04-27:
The RF24Network implementation currently retries failed fragments if a payload fails during sending.

The network layer will attempt up to 3 tries per failed fragment.

Since it also utilizes `writeFast()`, its not waiting for an ACK, just writing data to the FIFO, so if it fails, that means 3 payloads in the FIFO may fail, along with the current payload.

I propose adding logic that attempts to either:
1. Identify how many payloads are in the FIFO buffer after calling `txStandBy()` and re-send all failed payloads
2. Simply re-send the 4 payloads that may have failed

2bndy5, 2025-04-27:
Seems like 2 is the only option since 1 would incur extra overhead (trying to work with a FIFO as if it were a deque with an exact known capacity).

I thought 2 was the whole purpose of `txStandby()` due to using the indeterminable `writeFast()` (which could fail if FIFO is full or if a previous transmission failed). This would be easier to diagnose if RF24 exposed the STATUS flags without clearing them (`whatHappened()` may be inadequate).

I recently discovered that all Arduino cores use C++ std 11 as the default. This means we could explore using a [bit-field struct](https://en.cppreference.com/w/cpp/language/bit_field) to easily interpret the STATUS byte via RF24 API without allocating a whole byte for each boolean flag. I used [something similar in rf24-rs](https://docs.rs/rf24-rs/latest/rf24/struct.StatusFlags.html) as a mutable arg type to mask the IRQ pin (set), expose (get) the Status byte, and clear all or certain flags.

TMRh20, 2025-04-27:
It may be worthwhile to expose the status flags in RF24, otherwise I need to do wonky things like the following to get the exact state of the FIFO for RF24Network:
```cpp
bool RF24::txStandBy(uint32_t timeout, uint8_t *getFifo, bool startTx)
{

    if (startTx) {
        stopListening();
        ce(HIGH);
    }
    uint32_t start = millis();

    while (!(read_register(FIFO_STATUS) & _BV(TX_EMPTY))) {
        if (status & _BV(MAX_RT)) {
            uint8_t fifoStatus = isFifo(true);
            if(fifoStatus == 0){
                *getFifo = 1;
            }else
            if(fifoStatus == 1){
                *getFifo = 0;
            }else
            if(fifoStatus == 2){
                *getFifo = 3;
            }
            write_register(NRF_STATUS, _BV(MAX_RT));
            ce(LOW); // Set re-transmit
            ce(HIGH);
            if (millis() - start >= timeout) {
                ce(LOW);
                flush_tx();
                return 0;
            }
        }
#if defined(FAILURE_HANDLING) || defined(RF24_LINUX)
        if (millis() - start > (timeout + 95)) {
            errNotify();
    #if defined(FAILURE_HANDLING)
            return 0;
    #endif
        }
#endif
    }

    ce(LOW); //Set STANDBY-I mode
    return 1;
}
```

This would allow us to implement option 1, but as you can see adds a bit of overhead... I think having the STATUS flag exposed may be easier.

TMRh20, 2025-04-27:
The problem I'm finding with re-sending all potentially failed packets is that it often results in duplicate payloads, where finding the exact number of failed payloads is much more efficient.

2bndy5, 2025-04-27:
> The problem I'm finding with re-sending all potentially failed packets is that it often results in duplicate payloads

Sounds like a side effect of `reuseTX()`.

TMRh20, 2025-04-27:
Not sure, I don't get any duplicates after implementing it with option 1. I'm still playing around with that implementation to make sure its actually working correctly though.

2bndy5, 2025-04-27:
The patch from nRF24/RF24#1030?

2bndy5, 2025-04-27:
I created a RF24 branch named `expose-status-byte`. I know it compiles, but it isn't tested at all. (no python wrapper changes either)

TMRh20, 2025-04-27:
> The patch from https://github.com/nRF24/RF24/pull/1030?

No, I'm playing around with the code posted above to get the number of failed payloads during txStandBy();

I just realized it only tells you if its got 3, 0 or occupied (1 or 2) payloads in the FIFO. So `if(fifoStatus == 0){` should count as 2. 

Implementing this should theoretically make fragmented payloads send with greater reliability. How much is the question...

2bndy5, 2025-04-27:
> I just realized it only tells you if its got 3, 0 or occupied (1 or 2) payloads in the FIFO. So if(fifoStatus == 0){ should count as 2.

Using the returned [`rf24_fifo_state_e` enum](https://nrf24.github.io/RF24/group__fifoState.html) instead of the enumerated int ("magic") values would have helped you realize this quicker. 😛

TMRh20, 2025-04-27:
Hehehe probably

My testing results so far are inconclusive, I will leave this open for a while in case I come up with something that works.

2bndy5, 2025-04-27:
FWIW, this is how I'd modify the streamingData example:
```c++
    StatusFlags flags;
    unsigned int failures = 0; // keep track of failures
    uint8_t i = 0;
    clock_gettime(CLOCK_MONOTONIC_RAW, &startTimer); // start the timer
    while (i < SIZE) {
        makePayload(i);
        if (!radio.writeFast(&buffer, SIZE)) {
            radio.getStatusFlags(&flags);
            if (flags.tx_df) {
                failures++;
                // failed to transmit a previous payload.
                // Now we need to reset the tx_df flag and the CE pin
                radio.reUseTX(); // also spams the failed payload until writeFast() is called
            }
            // else the TX FIFO is full; just continue loop
        }
        else {
            i++;
        }

        if (failures >= 100) {
            // most likely no device is listening for the data stream
            cout << "Too many failures detected. ";
            cout << "Aborting at payload " << buffer[0];
            break;
        }
    } // while
```
Although, instead of using `reUseTX()`, I'd prefer to just
```cpp
radio.ce(LOW);
// where `flags.tx_df` is already asserted
radio.clearStatusFlags(flags);
radio.ce(HIGH);
```
That way, the TX FIFO isn't stuck spamming the failed payload until a new payload is uploaded (because that's what `reUseTX()` does).
Unfortunately `ce()` isn't public on the `expose-status-byte` branch.

BTW, in [`txStandby(timeout)`, there's a `flush_tx()` when the specified timeout is reached and the TX FIFO is not empty](https://github.com/nRF24/RF24/blob/170106bc6b7a7beb033fb0167dea303ca40b5c5a/RF24.cpp#L1466-L1476). This could lead to data loss...

TMRh20, 2025-04-27:
There is a flush_tx() in both `txStandBy()` functions, and yes it will lead to data loss, but only if the MAX_RT flag has been hit. At some point, you gotta give up and try again. 😄

TMRh20, 2025-05-07:
Been testing & thinking about this more and not sure I want to implement this unless also adding support for handling duplicate payloads. It could be a good method on Linux devices, which would typically use TCP/IP (RF24Gateway) for communication, but for Arduino devices, which can mix with Linux devices, it seems a bit overkill.
Re-sending the single currently lost, fragmented packet 3 times may also be a waste of time, but I haven't proven that out yet.

closed 2025-05-07 by TMRh20