nRF24/RF24Network #252: nRF52x is slightly too fast for nRF24 communication w/fragmented payloads

TMRh20, 2025-05-14:
I was doing some testing with RF24Network transfers between nRF52 & nRF24, and it seems to require a slight delay when sending fragmented packets from nRF52x to nRF24x:

I changed this:
https://github.com/nRF24/RF24Network/blob/aed6024c4e17555605082f50a87e91921b247f17/RF24Network.cpp#L783-L791

To this:
```cpp
        if (!ok) {
            delay(2);
            ++retriesPerFrag;
        }
        else {
            retriesPerFrag = 0;
            fragment_id--;
            msgCount++;
            delayMicroseconds(60);
        }
```

When sending 48-byte packets, the speed without the delay is about 10KB/s and around 32KB/s with the delay, not including the header.

Not sure how to approach this issue, because nRF52s work fine together without the delay, while the nRF24s require the delay, or else communication is hindered. The good thing is nRF52 supports much bigger payloads if communicating solely among other NRF52s, so the 60uS delay won't slow them down a huge amount if added.
Of course, if added, we should probably add a `#define` for this so advanced users can set it to 0 for nRF52 to nRF52 communication.

2bndy5, 2025-05-14:
I suppose we can use one of those unused bits in `networkFlags`. Set the bit to 1 if the delay is needed...

To set the bit automatically would be trickier though. It might require a separate implementation of the constructors for each supported radio type, similar to what we did recently for the `main_write()`.

2bndy5, 2025-05-14:
We could also just instruct users to set the bit themselves if the network is going to use a mix of radio types.

TMRh20, 2025-05-14:
Heh, I was just thinking of a blanket:

```cpp
#if defined NRF52_RADIO_LIBRARY
    delayMicroseconds(NRF52_FRAG_DELAY);
#endif
```

2bndy5, 2025-05-14:
But what if you use an nRF5x to drive a nRF24L while also participating in a separate network.

TMRh20, 2025-05-14:
Then you get a 60-65 uS delay. 😺

2bndy5, 2025-05-14:
It's lazy, but it would be less runtime computation than `if (networkFlags & SLOW_NRF5_FRAG)`

2bndy5, 2025-05-14:
> Of course, if added, we should probably add a `#define` for this so advanced users can set it to 0 for nRF52 to nRF52 communication.

Maybe something like
```c
#if !defined(THROTTLE_FRAG) && defined(NRF52_RADIO_LIBRARY)
    #define THROTTLE_FRAG 60
#else
    #define THROTTLE_FRAG 0
#endif

// then just test if the value is greater than 0
#if THROTTLE_FRAG > 0
    delayMicroseconds(THROTTLE_FRAG);
#endif
```

PR #253 (2025-05-15): Add delay during fragmentation for nRF52x
- nRF52x is slightly faster than nRF24, make up for this with a delay while sending fragmented payloads
- I increased the delay to 63uS since 60uS was the minimum that worked

Closes #252
closed 2025-05-15 by TMRh20